MichaelGrupp/evo #466: How to change the font type of all words to Times New Roman?

zxjhit, 2022-03-07:
Hello, first thank you for your great work for the slam community.
I want to change the font type to Times New Roman as the picture you show in [your plotting wiki](https://github.com/MichaelGrupp/evo/wiki/Plotting) like this,
[image]
However,when I use `evo_config set plot_fontfamily serif` and use evo to plot the trajectory,the plotting result is that the axis name(such as x,y,roll pitch yaw) font dosen't change at all,only the unit(such as m,deg) font changes,like this.
[image]
So how can I change the axis name font type to Times New Roman as well as the first picture shows?
Thank you for your reply.

MichaelGrupp, 2022-06-23:
The image in the Wiki is rendered with LaTeX. See https://github.com/MichaelGrupp/evo/wiki/Plotting#4-use-latex-renderer

The LaTeX font is not Times New Roman, by the way.

jeff210096, 2022-07-26:
Hello , i met this problem too. Is there any other way to change the front type into Time Tew Roman?

MichaelGrupp, 2023-01-11:
I recommend to check first if matplotlib supports this font. If it's possible with matplotlib somehow, then it's also possible with evo.

closed 2023-02-27 by MichaelGrupp