MichaelGrupp/evo #468: RPE when comparing identical trajectories

Kaapeine, 2022-04-13:
I was evaluating the RPE between a Ground Truth trajectory and a Visual Odometry trajectory when I noticed that the results printed in the console seemed off compared to the RPE vs Time plot. Here's what I mean: 

[image]

It didn't make sense that the mean is only 0.18 when half the values are close to 1.0. I was using this with the default delta i.e. 1. What could explain these results?

With delta = 30, I saw results that made more sense. 

[image]

Just to test out a hunch I had, I tried running evo_rpe comparing the same trajectory, while keeping delta = 1. I was surprised to see these results: 

[image]

This looks pretty similar to the first test which was between two different trajectories. Why am I seeing an RPE when I'm comparing the same trajectory files? Setting delta = 30 here gave me 0 RPE which is what I'd expected to see for delta = 1 as well. 

What's going wrong here? Is RPE supposed to be like this? I think there's something lacking in my understanding of how RPE works.

MichaelGrupp, 2022-04-13:
> Why am I seeing an RPE when I'm comparing the same trajectory files

I cannot reproduce this with the [test trajectories](https://github.com/MichaelGrupp/evo/tree/master/test/data), e.g. `evo_rpe kitti KITTI_00_gt.txt KITTI_00_gt.txt --delta 1`.

Without further information & data I can't say what's going wrong on your end.

Kaapeine, 2022-04-13:
I should've attached the trajectories, my bad. Here they are. 

[image]

[gt_mavros.txt](https://github.com/MichaelGrupp/evo/files/8479883/gt_mavros.txt)

MichaelGrupp, 2022-04-13:
Thanks, that makes it clearer. Your trajectory has multiple poses for the same timestamps. `evo_rpe/ape` will compare data at duplicated timestamps, even if it doesn't really make sense (I might change this in the future and throw an error / warning). You need to provide data with sorted, ascending timestamps to get meaningful results.

If you run `evo_traj tum gt_mavros.txt -p --full_check`, you can see that there's an error in the check and the plot has multiple data points at the same timestamp, i.e. jumps:
[image]

Kaapeine, 2022-04-13:
Thank you, I hadn't noticed this before. I was able to resolve it fixing the timestamps. The RPE went down by nearly a factor of 10, but there was no difference in the APE. Shouldn't there have been a difference with the APE as well?

MichaelGrupp, 2022-04-13:
No, because the absolute error doesn't take pairs from each trajectory and compares one-by-one. The problem in the RPE is that you get a relative pose pair with the same timestamp if there's a duplicated time and delta=1.

Kaapeine, 2022-04-13:
Ah that makes a lot of sense. Thank you so much for the quick support! Closing this now.

closed 2022-04-13 by Kaapeine