MichaelGrupp/evo #470: "evo_ape euroc" report [ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)

CobenGao, 2022-04-27:
**Description:**
when a try to use "evo_ape euroc" to get the APE of two Euroc format files,
it report error like below:
[DEBUG][2022-04-27 12:54:49,195][main_ape.run():255]
main_parser config:
{'align': True,
 'align_origin': False,
 'config': None,
 'correct_scale': False,
 'debug': True,
 'est_file': 'f_data.txt',
 'logfile': None,
 'n_to_align': -1,
 'no_warnings': False,
 'plot': True,
 'plot_colormap_max': None,
 'plot_colormap_max_percentile': None,
 'plot_colormap_min': None,
 'plot_full_ref': False,
 'plot_mode': 'xyz',
 'plot_x_dimension': 'seconds',
 'pose_relation': 'full',
 'ros_map_yaml': None,
 'save_plot': '.\\evo_result\\mono_inertial_euroc_ape',
 'save_results': '.\\evo_result\\mono_inertial_euroc_ape.zip',
 'serialize_plot': None,
 'silent': False,
 'state_gt_csv': 'data.csv',
 'subcommand': 'euroc',
 't_end': None,
 't_max_diff': 0.01,
 't_offset': 0.0,
 't_start': None,
 'verbose': True}
[DEBUG][2022-04-27 12:54:49,208][main_ape.run():256]
[DEBUG][2022-04-27 12:54:49,753][file_interface.read_euroc_csv_trajectory():209]
Loaded 28712 stamps and poses from: data.csv
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)

**Command:**
data.csv is the ./mav0/state_groundtruth_estimate0/data.csv of the  http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V2_01_easy/V1_01_easy.zip
f_data.txt looks like:
#Euroc format
#timestamp[ns],p_RS_R_x[m],p_RS_R_y[m],p_RS_R_z[m],q_RS_w[],q_RS_x[],q_RS_y[],q_RS_z[]
1403715278662142976.000000,0.025350310,-0.003238157,0.012935471,0.999806762,-0.004896020,0.018701242,0.003568859
1403715278712143104.000000,0.036598861,-0.003644791,0.018154707,0.999720454,-0.004097223,0.022476660,0.006094718
...

evo_ape euroc data.csv f_data.txt -r full -va --plot --plot_mode xyz --save_plot .\evo_result\mono_inertial_euroc_ape --save_results .\evo_result\mono_inertial_euroc_ape.zip --debug
```

CobenGao, 2022-04-27:
may the est_file must be TUM format, I change it as the below format, it works
#TUM format
#timestamp[s] p_RS_R_x[m] p_RS_R_y[m] p_RS_R_z[m] q_RS_x[] q_RS_y[] q_RS_z[] q_RS_w[]
1403715278.662143 -0.003252760 -0.026221097 0.015050826 0.018544139 0.005165383 0.004063323 0.999806464
1403715278.712143 -0.003765939 -0.038424030 0.020605311 0.022264620 0.004432760 0.006688863 0.999719977

closed 2022-06-23 by MichaelGrupp

GeJintian, 2023-09-18:
> As you wrote, the other file needs to be in TUM format. If you have an euroc format file you can convert it using `evo_traj`, see the file format section of the Wiki.

@MichaelGrupp Hello, so you mean that at least one of the file must be in TUM format, even if we use "evo_ape euroc" command line?