MichaelGrupp/evo #471: Readjust axis limits

s-a-n-d-y, 2022-05-06:
Hi!

First of all, thank you for making this wonderful tool public. 

**I have a question:** Is it possible to change the range of the y-axis. For example, in the picture below you could see that the yaw of the blue line has a similar phase as the magenta. However, probably near t=50, the yaw for the blue line didn't rotate as much as the magenta. Hence, we see an abrupt jump on the magenta (perhaps the yaw became more than 180).
 
[image]

Since the limit on the yaw scale is from -180 -- +180, it looks like things are out of phase. However, this is more of an issue in terms of visualization which can be fixed if the range is increased from -360 -- +360. Are there any settings in the tool where this may be tweaked?

MichaelGrupp, 2022-06-23:
This angle is normalized to [-pi, pi], in degrees as shown in the plot it's [-180°, 180°]. Keep in mind that an angle difference is not an absolute difference of the number values - e.g. the angle difference between 179° and -179° is 2° and not 358°.

> which can be fixed if the range is increased from -360 -- +360
Which will look the same if something spins further than one rotation - there's just no way to display an angle normalized to an interval without ever having discontinuities in the pure number value.

closed 2022-07-28 by MichaelGrupp

s-a-n-d-y, 2022-08-29:
Thanks for your reply. I understand what you mean. Maybe I will tweak the visualization for my specific case to make it look nicer.