MichaelGrupp/evo #486: Doubt about evo_ape result

belensaacastro, 2022-07-02:
Hi! I'm plotting ground truth and poses exported from rtabmap database viewer. 

When I plot with evo_traj it looks ok:  
[image]

But evo_ape returns me this weird plot, can you help me please?
[image]

Cheers

MichaelGrupp, 2022-07-04:
You need to provide your data and which full command you are running for others to be able to reproduce what you are seeing.

belensaacastro, 2022-07-05:
Hi! 
I'm using multisession rtabmap. I concatenated both ground truth tum file into one.  I want to comapre it with the rtabmap's result on the same plot.

I run this command:
evo_ape tum mediodia.tum poses_rgbd.txt --align --plot.

I've tried on the jupyter notebook documentation example, synchronization seems to be ok, but the plot look the same as the above one.

You can find the files here: https://drive.google.com/drive/folders/1eLECqYii_OUbH1yXhL79lDyhpSbHRoWX

Thanks!

MichaelGrupp, 2022-07-05:
Your files are broken, the `mediodia.tum` and `poses_rgbd.txt` both have unsorted timestamps. They have a jump back in time, which explains the weird looking plot because the compared poses are associated by timestamp in evo_ape. This can be also seen if you use the xyz view of `evo_traj`. You can also see if a file is ok by adding the `--full_check` option to evo_traj.

belensaacastro, 2022-07-07:
Hi! Thank you! 

I got metiodia.tum concatenating mediodia_a.tum and mediodia_b.tum. I got both of these files with the command 
`evo_traj bag mediodia_a.bag ground_truth_topic --save_as_tum`
I believe this command only takes trajectory from odom topic. Which timestamps are being assigned?

I'm playing the bags using --clock and on rtabmap I configured rosparam use_sim_time on true. Maybe thats why the resulting poses are also jumping back in time, make sense? 

Cheers,

belensaacastro, 2022-07-17:
Hi, I figured out the problem about the timestamps. When running the simulator to record the bags, the time restarted again, that's why the timestamps were mixed; and setting the rosparm use_sim_time to false did not fix that.

Thanks for your time!
Cheers

closed 2022-07-17 by belensaacastro