MichaelGrupp/evo #490: KITTI map same but differnt oientation when plotted

col761-22, 2022-08-24:
Hi,
 Thanks for this awesome project. Really helped me a lot.

I had a question, rather a bug that I am not able to solve.

I used groundtruth ktti poses and converted it to tum format. Further, I used cartogrpaher to run SLAM on a sequence 005(2011_09_30 - 0018). I have to calcualte the error between both the poses for my research.

However when I plot both the pose maps, they have different orientation, although the overall structure is the same.
Here is the glimpse of the same.
[image]

I am not able to align it, which is required for finding ate/rpe between the poses.

The trajectory txt required to plot this image is as followd:
[kitti05.txt](https://github.com/MichaelGrupp/evo/files/9417122/kitti05.txt)
 &
[kitti_2011_09_30_drive_0018_synced.bag.txt](https://github.com/MichaelGrupp/evo/files/9417126/kitti_2011_09_30_drive_0018_synced.bag.txt)

It woulbe great if someone could throw light on this and help me figure out the correct way to aligb both the trajectories.

eithwa, 2022-08-25:
```
$ evo_traj tum kitti05.txt --ref  kitti_2011_09_30_drive_0018_synced.bag.txtp  -as --t_offset +1317363879.336294
```
[image]
https://github.com/MichaelGrupp/evo/wiki/evo_traj#geometric-alignment

col761-22, 2022-08-26:
Thanks a lot. This helped me a lot

col761-22, 2022-09-08:
Hi, I had one more issue.

Currently both the files The trajectory txt required to plot this image is as followd:
[kitti05.txt](https://github.com/MichaelGrupp/evo/files/9417122/kitti05.txt)
&
[kitti_2011_09_30_drive_0018_synced.bag.txt](https://github.com/MichaelGrupp/evo/files/9417126/kitti_2011_09_30_drive_0018_synced.bag.txt) cannot be used for calcualting ate/rpe because they are in different planes.

How do I convert the  kitti_2011_09_30_drive_0018_synced.bag.txt file in the same plane ane then evaluate ape/rpe numbers..

I understand that visulaization is in the same plane. But currently the data in files are in diffrent planes. 
Request your urgent help on this.

Thanks in advance.

prashkmr, 2022-09-10:
Hi 
I wanted to know that how did you come up with this time t_offset value in the above error.

eithwa, 2022-09-16:
> Hi I wanted to know that how did you come up with this time t_offset value in the above error.

The time stamp of the first data of kitti05.txt is 0.00000000000000000e+00, and the time stamp of the first data of kitti_2011_09_30_drive_0018_synced.bag.txt is 1317363879.336294. Use --t_offset +1317363879.336294 to align the time stamps on both sides

eithwa, 2022-09-16:
> 

> Hi, I had one more issue.
>
> Currently both the files The trajectory txt required to plot this image is as followd: [kitti05.txt](https://github.com/MichaelGrupp/evo/files/9417122/kitti05.txt) & [...]

you can try to convert its xyz axis using python and import pandas

closed 2022-12-01 by MichaelGrupp