MichaelGrupp/evo #496: How to find the ape error reason?

JACKLiuDay, 2022-09-20:
Hi, thank you for your great about evo. I want to test my lidar localization error with RTK. And I got a traj with my lidar about 2468 data. And my RTK data number is 2492. But when I run ''evo_ape tum 20220916dog_local_in.txt gardeninrtk.txt --plot_mode xy -p -a -s --save_results test1.zip''. I got ape arror number is 194. Shouldn't the number of errors compared be 2468？
[image]
I want to analyze the error data with imu data. In this way, the influence of motion on positioning error can be analyzed. But these data obtained by the evo make it difficult for me to carry out the next step.

MichaelGrupp, 2022-09-20:
Hard to say anything without knowing your data, but I would assume that there are just 194 matches found between the timestamps. Run with `--verbose` to see what's happening.

JACKLiuDay, 2022-09-20:
> Hard to say anything without knowing your data, but I would assume that there are just 194 matches found between the timestamps. Run with `--verbose` to see what's happening.

Hi, thank you for your quick reply. I will upload my lidar pose data and RTK data soon.

JACKLiuDay, 2022-09-21:
> > Hard to say anything without knowing your data, but I would assume that there are just 194 matches found between the timestamps. Run with `--verbose` to see what's happening.
> 
> Hi, thank you for your quick reply. I will upload my lidar pose data and RTK data soon. [image] [image] [image] It looks like my time stamp has some problem. Is it because the timestamps of my radar data and RTK timestamps are not synchronized?
[image]
Hi, I found something wrong with my RTK timestamp. I corrected it and test it with evo. Now we have 1007 match pose.