MichaelGrupp/evo #497: When using evo ape, do we need to compensate for installed external parameters between GPS and lidar odometry?

JACKLiuDay, 2022-09-22:
Hi, thank you for your great work about evo. When I running my robot with lidar and GPS, I have some questions. My GPS and lidar are mounted at different places. There is a nearly 20cm installation position deviation between them. When I use evo_ape, do I need to compensate for the deviation of  the installation position first?  Can evo_ape compensate for the translation of this installation position through the -a and -s parameters?

MichaelGrupp, 2022-09-23:
This is an extrinisc calibration that you need to solve. The alignment methods in evo cannot do this. They assume that the trajectories were captured by the same sensor frame. If your sensors are not in the same position, this doesn't hold and you would get different trajectory shapes e.g. when you rotate your platform.

JACKLiuDay, 2022-09-24:
> This is an extrinisc calibration that you need to solve. The alignment methods in evo cannot do this. They assume that the trajectories were captured by the same sensor frame. If your sensors are [...]

Hi,thank you so much for your quick reply. This problem has troubled me for a long time.  The error I get from evo_ape analysis is always very strange. Now I will compensate the external parameters of RTK installation and re-experiment. Thank you!

closed 2022-09-24 by JACKLiuDay