MichaelGrupp/evo #505: some propblems of tum traj align

MoonSheepJx, 2022-11-21:
Hello, i have some problems about align traj in tum format!
I use evo to align groundtruth(tum) from RTK and traj(tum) from my output,
before align them, i see the traj of them is them is normal, but when i align them(either basis), one of the traj becomes distorted.
as below:
[image]

but after align:
[image]

my datas info is  
[image]

and my command is 
`evo_traj tum 20190117.tum fast_lio.txt -a --ref=20190117.tum --save_as_tum`

my data is 

[20190117.txt](https://
[fast_lio.txt](https://github.com/MichaelGrupp/evo/files/10051930/fast_lio.txt)
github.com/MichaelGrupp/evo/files/10051929/20190117.txt)
(fast-lio is txt, but i changed the 20190117.tum suffix from .tum to .txt for upload to github!)
thanks for any help!

MichaelGrupp, 2022-11-21:
Only a few poses can be matched with the timestamps. Run with `--verbose` and you will see:
```
Found 60 of max. 318 possible matching timestamps between...
	reference
and:	fast_lio.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).

```

With a larger `--t_max_diff 0.05`, there are more matches:

```
Found 318 of max. 318 possible matching timestamps between...
	reference
and:	fast_lio.txt
..with max. time diff.: 0.05 (s) and time offset: 0.0 (s).
```

You will need to check yourself if this is correct or if there is an error in your dataset.

MoonSheepJx, 2022-11-21:
I really appreciate your quick reply, which will help me a lot.
As you mentioned, only a few poses can be matched.
The ground truth of my dataset is from RTK (1Hz), so poses is small compared with the LiDAR frames.

I'm not sure if the low number of ground true frames is causing this problem, I would like to get your opinion.
(From the traj of each file, all are correct, and I also know from the publisher of the dataset that the truth value can only be obtained at a frequency of 1hz)
And for this problem, if i can not add the frame of ground truth, is there any way to adjust it other than using the `time offset`

Thanks again for your reply, which will be of great help to me. I am looking forward to your reply and suggestions！
Thank you！

MoonSheepJx, 2022-11-21:
really appreciate your help.
 I have carried out the test according to your method, and the result is seem good. Thank you very much.
[image]