MichaelGrupp/evo #506: Statistics plots do not convey a correct meaning! box-pot is not True.

Delbina, 2022-11-23:
Hi,
I have plotted the APE-statistics, but something looks strange. according to the barchart, the maximum value is 4, but in the box plot, it shows around 1.3.
[image]

according to the definition of box-plot, the lower line is minimum, the upper line represents maximum, and the line in the middle is median. Those dots are outliers. so it seems box-plot is not True!
How to solve this issue

MichaelGrupp, 2022-11-23:
There are just different ways to draw a box plot. https://en.wikipedia.org/wiki/Box_plot#Elements

The one you see here plots your max values as outliers.

Delbina, 2022-11-23:
thanks for your comment,
but even in the link you shared with me, it is mentioned: 
[Maximum](https://en.wikipedia.org/wiki/Sample_maximum) (Q4 or 100th percentile): the highest data point in the data set excluding any outliers

Would you please send me the definition of the box-plot defined in this package?
If there is any document that explain all the results of ape-evo (tables, graphs,...), I would be grateful if you send it to me.

Kind Regards

MichaelGrupp, 2022-11-23:
"The whiskers can be defined in various ways."

The box plot that evo uses is: https://seaborn.pydata.org/generated/seaborn.boxplot.html

Delbina, 2022-11-24:
based on your comment, now i should mention in my report that those dots are the maximum value??
the upper line should represent the max value, so the line should cover those points as well.