MichaelGrupp/evo #515: A large part of the trajectory is missing after aligning the trajectory

chenhuahua-orz, 2022-12-08:
Hi, first of all, thank you very much for bringing such an excellent software!
I found some problems while evaluating the trajectory of LIO-SAM using EVO.
These are the two tracks where I did not use -a,You can see that the two trajectories are complete.
[image]

After I aligned the trajectory, I found a large part of the trajectory was missing.
[image]
[image]

Here are my two track files,I hope you can help me to check what is wrong.
[gt_slam_UrbanNav-HK-TST-20210517.csv](https://github.com/MichaelGrupp/evo/files/10184543/gt_slam_UrbanNav-HK-TST-20210517.csv)
[transformations.txt](https://github.com/MichaelGrupp/evo/files/10184545/transformations.txt)

Hope you have a great day!

MichaelGrupp, 2022-12-08:
This just means that some of your timestamps can't be matched.

Please take a look at my comment in a different issue where someone had the same question: https://github.com/MichaelGrupp/evo/issues/505#issuecomment-1321868473

chenhuahua-orz, 2022-12-09:
> This just means that some of your timestamps can't be matched.
> 
> Please take a look at my comment in a different issue where someone had the same question: [#505 (comment)](https://github.com/MichaelGrupp/evo/issues/505#issuecomment-1321868473)

Thank you！！

closed 2023-01-11 by MichaelGrupp