MichaelGrupp/evo #523: How can we use evo to plot trajectory with real-time data update?

nitik1998, 2022-12-20:
Hi, 
So I am working on a monocular odometery project, where I calculate the trajectory with real time video input. Right now I'm saving the trajectory file in KITTI pose format and then visualising the trajectory using `evo_traj`. Is there any way I can use evo to plot and update trajectory in real time?

MichaelGrupp, 2022-12-21:
evo is not designed for live visualization, and adding such a functionality would require major fundamental changes in the software.

I would rather recommend to look into one of the common, already existing live visualization solutions like RViz, Foxglove etc or visualization libraries like Pangolin.

nitik1998, 2022-12-21:
Thanks @MichaelGrupp !

closed 2022-12-21 by nitik1998