MichaelGrupp/evo #524: Set plot_pose_correspondences from config.json

Alex-Beh, 2023-01-12:
If I understand correctly, no all the parameters can be set via the config file `--config`.

For example, I would like to set `plot_pose_correspondences` parameter via the config and it don't work. The only way I can do is via `evo_config set plot_pose_correspondences true` which change the value in the default_path.

May I know do I understand correctly? Else how can I set the `plot_pose_correspondences` via config file.

reference:
1. https://github.com/MichaelGrupp/evo/blob/019cdb8bbc3730a2e1cf5bfb99e6e1d9de2deaa5/evo/common_ape_rpe.py#L172
2. https://github.com/MichaelGrupp/evo/blob/019cdb8bbc3730a2e1cf5bfb99e6e1d9de2deaa5/evo/tools/settings.py#L34

Alex-Beh, 2023-01-12:
Tested the `override-settings-via-cli` branch and it works incredibly. 

I will close the issue after you merged the branch~~ Thanks for the fast response.

MichaelGrupp, 2023-01-12:
Correct, such global parameters could not be set from the `--config` file yet, only command line flags. But since this can be useful and it's easy to add this feature, I implemented it here: #525 

With this you can specify a custom value for parameters like `plot_pose_correspondences` in the json file together with command line flags and it will only be used for that execution, or have a custom copy of the `settings.json` etc.

closed 2023-01-12 by Alex-Beh