MichaelGrupp/evo #526: why on the plots related error mapped onto trajectories, the references are not the same.?

Delbina, 2023-01-15:
Hi 
I have considered GPS as my ground truth, but I do not know why on the plots related error mapped onto trajectories, the references are not the same.
[image]

thanks

MichaelGrupp, 2023-01-19:
Add `--verbose` to your command and check the output how many poses can be matched in each case.

Probably that number is lower in the right plot.

Delbina, 2023-01-19:
Would you please explain more? I did not get your point!
I have compared to trajectories with the same ground truth, why the ground truth is different?

MichaelGrupp, 2023-01-19:
The ground truth is not different, probably just different poses are used for the comparison because of matching.
Poses in ground truth are matched to the poses in your trajectory by matching the timestamps. If less timestamps can be matched or there are less poses in general, the number of poses will vary. You can see this in the output when you add `--verbose`.

Delbina, 2023-01-19:
> The ground truth is not different, probably just different poses are used for the comparison because of matching. Poses in ground truth are matched to the poses in your trajectory by matching the [...]

Interesting, Thanks a lot

closed 2023-02-27 by MichaelGrupp