MichaelGrupp/evo #527: evo commands not found after install ubuntu

SashelI, 2023-01-16:
**Description:**

Hi, i'm trying to install evo on an Ubuntu 18.04.3 VM using python 3.8.16.
The installation goes fine, but when I try to invoke "evo_config" or any evo command, it says (in french) "command nor found". even on a new terminal.

**Command:**
```
python3.8 -m pip install evo --upgrade --no-binary evo

evo_config
```

**Console output:**
```

robot@goose:~$ python3.8 -m pip install evo --upgrade --no-binary evo
Defaulting to user installation because normal site-packages is not writeable
DEPRECATION: --no-binary currently disables reading from the cache of locally built wheels. In the future --no-binary will not influence the wheel cache. pip 23.1 will enforce this behaviour change. A possible replacement is to use the --no-cache-dir option. You can use the flag --use-feature=no-binary-enable-wheel-cache to test the upcoming behaviour. Discussion can be found at https://github.com/pypa/pip/issues/11453
Collecting evo
  Using cached evo-1.21.0.tar.gz (121 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy>=1.18.5 in ./.local/lib/python3.8/site-packages (from evo) (1.24.1)
Collecting matplotlib
  Downloading matplotlib-3.6.3-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (9.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 9.4/9.4 MB 38.1 MB/s eta 0:00:00
Collecting scipy>=1.2
  Downloading scipy-1.10.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (34.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 34.5/34.5 MB 11.7 MB/s eta 0:00:00
Collecting pandas
  Downloading pandas-1.5.2-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (12.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.2/12.2 MB 40.5 MB/s eta 0:00:00
Collecting numexpr>=2.7.3
  Downloading numexpr-2.8.4-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (381 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 381.7/381.7 kB 23.8 MB/s eta 0:00:00
Collecting seaborn>=0.9
  Downloading seaborn-0.12.2-py3-none-any.whl (293 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 293.3/293.3 kB 47.1 MB/s eta 0:00:00
Collecting natsort
  Using cached natsort-8.2.0-py3-none-any.whl (37 kB)
Collecting argcomplete
  Using cached argcomplete-2.0.0-py2.py3-none-any.whl (37 kB)
Collecting colorama>=0.3
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting pygments
  Using cached Pygments-2.14.0-py3-none-any.whl (1.1 MB)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from evo) (3.12)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from evo) (5.1.0)
Requirement already satisfied: rosbags>=0.9.10 in ./.local/lib/python3.8/site-packages (from evo) (0.9.14)
Requirement already satisfied: zstandard in ./.local/lib/python3.8/site-packages (from rosbags>=0.9.10->evo) (0.19.0)
Requirement already satisfied: ruamel.yaml in ./.local/lib/python3.8/site-packages (from rosbags>=0.9.10->evo) (0.17.21)
Requirement already satisfied: lz4 in ./.local/lib/python3.8/site-packages (from rosbags>=0.9.10->evo) (4.3.2)
Collecting python-dateutil>=2.7
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 247.7/247.7 kB 40.5 MB/s eta 0:00:00
Collecting kiwisolver>=1.0.1
  Downloading kiwisolver-1.4.4-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 29.1 MB/s eta 0:00:00
Collecting contourpy>=1.0.1
  Downloading contourpy-1.0.7-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (300 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 300.0/300.0 kB 17.5 MB/s eta 0:00:00
Collecting pyparsing>=2.2.1
  Downloading pyparsing-3.0.9-py3-none-any.whl (98 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 98.3/98.3 kB 37.9 MB/s eta 0:00:00
Collecting packaging>=20.0
  Downloading packaging-23.0-py3-none-any.whl (42 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.7/42.7 kB 17.6 MB/s eta 0:00:00
Collecting pillow
  Downloading Pillow-9.4.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.3/3.3 MB 35.2 MB/s eta 0:00:00
Collecting fonttools>=4.22.0
  Downloading fonttools-4.38.0-py3-none-any.whl (965 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 965.4/965.4 kB 25.4 MB/s eta 0:00:00
Collecting cycler>=0.10
  Downloading cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting pytz>=2020.1
  Downloading pytz-2022.7.1-py2.py3-none-any.whl (499 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 499.4/499.4 kB 61.6 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.7->matplotlib->evo) (1.11.0)
Requirement already satisfied: ruamel.yaml.clib>=0.2.6 in ./.local/lib/python3.8/site-packages (from ruamel.yaml->rosbags>=0.9.10->evo) (0.2.7)
Installing collected packages: pytz, scipy, python-dateutil, pyparsing, pygments, pillow, packaging, numexpr, natsort, kiwisolver, fonttools, cycler, contourpy, colorama, argcomplete, pandas, matplotlib, seaborn, evo
  WARNING: The script pygmentize is installed in '/home/robot/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script natsort is installed in '/home/robot/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts fonttools, pyftmerge, pyftsubset and ttx are installed in '/home/robot/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  DEPRECATION: evo is being installed using the legacy 'setup.py install' method, because the '--no-binary' option was enabled for it and this currently disables local wheel building for projects that don't have a 'pyproject.toml' file. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/11451
  Running setup.py install for evo ... done
Successfully installed argcomplete-2.0.0 colorama-0.4.6 contourpy-1.0.7 cycler-0.11.0 evo-1.21.0 fonttools-4.38.0 kiwisolver-1.4.4 matplotlib-3.6.3 natsort-8.2.0 numexpr-2.8.4 packaging-23.0 pandas-1.5.2 pillow-9.4.0 pygments-2.14.0 pyparsing-3.0.9 python-dateutil-2.8.2 pytz-2022.7.1 scipy-1.10.0 seaborn-0.12.2
robot@goose:~$ evo_config
evo_config : commande introuvable
robot@goose:~$ evo_config --help
evo_config : commande introuvable

```

* evo version number  : 

robot@goose:~$ evo pkg --version
La commande « evo » n'a pas été trouvée

* didnt change the source code.
```
```

MichaelGrupp, 2023-01-17:
The log output shows that pip did a user installation.
The executables are installed in your user folder and should show up when you run `ls ~/.local/bin/evo*`.

If they exist there and you can't execute them, then probably this `~/.local/bin` folder is not in your `PATH` variable. By adding this line to your shell startup file (`~/.bashrc`) it should work in a new terminal:
```sh
export PATH="$HOME/.local/bin:$PATH"
```

SashelI, 2023-01-18:
> The log output shows that pip did a user installation. The executables are installed in your user folder and should show up when you run `ls ~/.local/bin/evo*`.
> 
> If they exist there and you can't execute them, then probably this `~/.local/bin` folder is not in your `PATH` variable. By adding this line to your shell startup file (`~/.bashrc`) it should work in a new terminal:
> 
> ```shell
> export PATH="$HOME/.local/bin:$PATH"
> ```

Thanks, I indeed need to execute the commands locally, in ./local/bin.

closed 2023-01-18 by SashelI