MichaelGrupp/evo #528: Rosbag TF axis error

SashelI, 2023-01-18:
**Description:**

Hi (again), i've got a similar issue from this one that was closed without any explanation : when I try to evo_traj, i got the error "axis 1 is out of bounds for array of dimension 1" with my /tf topic.

**Command:**
```
evo_traj bag test_evo_rec2.bag /orb_slam3/camera_pose /tf:world.camera
```

**Console output:**
```

--------------------------------------------------------------------------------
name:	/orb_slam3/camera_pose
infos:	948 poses, 20.864m path length, 62.644s duration
--------------------------------------------------------------------------------
name:	/tf:world.camera
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/robot/.local/lib/python2.7/site-packages/evo/entry_points.py", line 98, in launch
    main_module.run(args)
  File "/home/robot/.local/lib/python2.7/site-packages/evo/main_traj.py", line 374, in run
    print_compact_name)
  File "/home/robot/.local/lib/python2.7/site-packages/evo/main_traj.py", line 274, in print_traj_info
    logger.info("infos:\t" + str(traj))
  File "/home/robot/.local/lib/python2.7/site-packages/evo/core/trajectory.py", line 247, in __str__
    s = super(PoseTrajectory3D, self).__str__()
  File "/home/robot/.local/lib/python2.7/site-packages/evo/core/trajectory.py", line 67, in __str__
    self.path_length)
  File "/home/robot/.local/lib/python2.7/site-packages/evo/core/trajectory.py", line 141, in path_length
    return float(geometry.arc_len(self.positions_xyz))
  File "/home/robot/.local/lib/python2.7/site-packages/evo/core/geometry.py", line 88, in arc_len
    return np.sum(np.linalg.norm(x[:-1] - x[1:], axis=1))
  File "/home/robot/.local/lib/python2.7/site-packages/numpy/linalg/linalg.py", line 2481, in norm
    return sqrt(add.reduce(s, axis=axis, keepdims=keepdims))
AxisError: axis 1 is out of bounds for array of dimension 1

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

```

**Additional files:**
This is my .bag file : [test_evo_rec2.bag](https://drive.google.com/file/d/1rPC_3XzYlF6kU_CRipZDSVEQpd9FrnQT/view?usp=share_link)

and this is my /tf topic :
```
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1671108949
        nsecs: 299504280
      frame_id: "world"
    child_frame_id: "camera"
    transform: 
      translation: 
        x: -0.278285384178
        y: -0.308478444815
        z: -0.0703073069453
      rotation: 
        x: 0.00721352714115
        y: -0.0290671772772
        z: 0.0501639558304
        w: 0.998291866445
---

```

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```
evo version : v1.12.0
python version : 2.7.17
operating system : Ubuntu 18.04, ROS melodic
didnt change the source code

evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "TkAgg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_axis_marker_scale": 0.0, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_cmap": "Greys_r", 
    "ros_map_unknown_cell_value": 205, 
    "ros_map_viewport": "keep_unchanged", 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2023-01-18:
I think that's because the TF trajectory only has a single pose and the old v1.12.0 release for Python 2 that you are using with ROS melodic didn't seem to handle this correctly.

I can't test or provide a fix here because I don't support Python 2 anymore.

With the latest version with Python 3:
```
$ evo_traj bag test_evo_rec2.bag /orb_slam3/camera_pose /tf:world.camera
--------------------------------------------------------------------------------
name:	/orb_slam3/camera_pose
infos:	948 poses, 20.864m path length, 62.644s duration
--------------------------------------------------------------------------------
name:	/tf:world.camera
infos:	1 poses, 0.000m path length, 0.000s duration
```

Maybe it's possible to do a workaround by trying to incorporate some changes that are in newer versions, like e.g.:
* https://github.com/MichaelGrupp/evo/commit/0b369a4e9532f2239d53f6be352aae50a3f9a50f
* https://github.com/MichaelGrupp/evo/commit/051e5bf63195172af58dc8256cc71618f079f224

SashelI, 2023-01-18:
> I think that's because the TF trajectory only has a single pose and the old v1.12.0 release for Python 2 that you are using with ROS melodic didn't seem to handle this correctly.
> 
> I can't test or provide a fix here because I don't support Python 2 anymore.
> 
> With the latest version with Python 3:
> 
> ```
> $ evo_traj bag test_evo_rec2.bag /orb_slam3/camera_pose /tf:world.camera
> --------------------------------------------------------------------------------
> name:	/orb_slam3/camera_pose
> infos:	948 poses, 20.864m path length, 62.644s duration
> --------------------------------------------------------------------------------
> name:	/tf:world.camera
> infos:	1 poses, 0.000m path length, 0.000s duration
> ```
> 
> Maybe it's possible to do a workaround by trying to incorporate some changes that are in newer versions, like e.g.:
> 
> * [0b369a4](https://github.com/MichaelGrupp/evo/commit/0b369a4e9532f2239d53f6be352aae50a3f9a50f)
> * [051e5bf](https://github.com/MichaelGrupp/evo/commit/051e5bf63195172af58dc8256cc71618f079f224)

thanks. For compatibility issues I can't update to newer ROS, and thus can't go further than pyhton 2.7.
I'll try the workarounds.

closed 2023-02-27 by MichaelGrupp