MichaelGrupp/evo #532: --ros_map_yaml unhandled error

mrtnbm, 2023-02-23:
**Description:**
- When adding --ros_map_yaml an unhandled error is thrown

**Command:**
```
evo_traj tum pose.txt -p --plot_mode xy --ros_map_yaml map.yaml 
```

**Console output:**
```

name:	dots_marker_night_very_slow_slam
infos:	147 poses, 9.225m path length, 92.301s duration
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/x/.local/lib/python3.8/site-packages/evo/entry_points.py", line 99, in launch
    main_module.run(args)
  File "/home/x/.local/lib/python3.8/site-packages/evo/main_traj.py", line 505, in run
    plot.ros_map(ax_traj, args.ros_map_yaml, plot_mode)
  File "/home/x/.local/lib/python3.8/site-packages/evo/tools/plot.py", line 721, in ros_map
    original_bbox = copy.deepcopy(ax.dataLim)
  File "/usr/lib/python3.8/copy.py", line 153, in deepcopy
    y = copier(memo)
  File "/usr/lib/python3/dist-packages/matplotlib/transforms.py", line 121, in __copy__
    raise NotImplementedError(
NotImplementedError: TransformNode instances can not be copied. Consider using frozen() instead.

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```
evo pkg --version v1.21.0
Ubuntu 20.04.5
no
evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        5,
        5
    ],
    "plot_fontfamily": "serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 0.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "whitegrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "zoom_to_map",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```
[poses.txt](https://github.com/MichaelGrupp/evo/files/10809540/poses.txt)
Cant attach the yaml and pgm files as github does not support these file types.

MichaelGrupp, 2023-02-24:
Which version of matplotlib are you using?

mrtnbm, 2023-02-24:
Thank you for the reminder to check the matplotlib version. It was 2 years old and therefore some new changes were missing.

closed 2023-02-24 by mrtnbm