MichaelGrupp/evo #534: ImportError: Cannot load backend 'Qt5Agg' which requires the 'qt5' interactive framework, as 'headless' is currently running

guoguoyi, 2023-02-25:
**Description:**
I have an import error when running this evo on Linux:

**Console output:**
Traceback (most recent call last):
  File "/home/hsr/UnVIO-master/evo-master/evo/entry_points.py", line 99, in launch
    main_module.run(args)
  File "/home/hsr/UnVIO-master/evo-master/evo/main_ape.py", line 294, in run
    common.plot_result(args, result, traj_ref,
  File "/home/hsr/UnVIO-master/evo-master/evo/common_ape_rpe.py", line 113, in plot_result
    from evo.tools import plot
  File "/home/hsr/UnVIO-master/evo-master/evo/tools/plot.py", line 35, in <module>
    import matplotlib.pyplot as plt
  File "/home/hsr/anaconda3/lib/python3.9/site-packages/matplotlib/pyplot.py", line 2500, in <module>
    switch_backend(rcParams["backend"])
  File "/home/hsr/anaconda3/lib/python3.9/site-packages/matplotlib/pyplot.py", line 285, in switch_backend
    raise ImportError(
ImportError: Cannot load backend 'Qt5Agg' which requires the 'qt5' interactive framework, as 'headless' is currently running

[ERROR] evo module evo.main_ape crashed - no logfile written (disabled)

```

MichaelGrupp, 2023-02-27:
Try `evo_config set plot_backend tkagg` or `evo_config set plot_backend agg`. For any further help you need to describe a bit better what you are doing.

guoguoyi, 2023-03-08:
Thanks a lot!! I tried "evo_config set plot_backend tkagg", it still didn't work. But setting "evo_config set plot_backend agg" is useful and the problem has been sovled

closed 2023-04-20 by MichaelGrupp