MichaelGrupp/evo #537: Compute error along one axis

fratopa, 2023-03-09:
Hello, I just discovered this project and it seems to be exactly what I am looking for. So far I have been able to compute the odometry errors and produce plots of them but I would like to restrict the error computations along a single axis. Is that possible?
Specifically, I would like to be able to plot the error values over time along the Z axis.
If someone knows how to do it or point me to where it is explained in the documentation I would really appreciate it

1475015695, 2023-04-20:
cool, thats what I want too.

MichaelGrupp, 2023-07-14:
Similar question asked in #554, see my comment there for a detailed answer: https://github.com/MichaelGrupp/evo/issues/554#issuecomment-1636028139

If you want to use only Z and don't want to write a script, you could also consider setting all x & y values in your input to 0 as a quick hack.

closed 2023-07-14 by MichaelGrupp