MichaelGrupp/evo #543: ATE and RTE

an99990, 2023-04-14:
**Description:**
Hi, So i currently have different trajectories on different reference frames and would like to calculate the error between them. I would like to know if theres a way to modifiy the current evo_rpe/evo_ape to get the trajectories instead.

Thank you

MichaelGrupp, 2023-04-17:
I don't understand what you mean with "get the trajectories instead".

If you have data in different frames, you can try to do an [alignment](https://github.com/MichaelGrupp/evo/wiki/Metrics#alignment) step in evo.

an99990, 2023-04-17:
Hi @MichaelGrupp Thank you for your answer,
is there a way to calculate the ATE with this package or is there a way i can modify the code to get that metric ?

MichaelGrupp, 2023-04-17:
Running `evo_ape` with its default translation pose relation and alignment enabled is calculating "ATE", if you mean absolute trajectory error as per definition of Sturm et al. for example.

MichaelGrupp, 2023-04-18:
This means that there is too little information to compute the alignment. If your data has less than 3 non-colinear points, this rank-deficiency in the Umeyama alignment will happen. See also this PR where we discussed this topic a bit more in depth: https://github.com/MichaelGrupp/evo/pull/240

closed 2023-06-26 by MichaelGrupp