MichaelGrupp/evo #546: EVO tool is not aligning the origin of two trajectories ground truth trajectory and slam trajectory.

sagardhatrak, 2023-05-04:
**Description:**

**Command:**
evo_ape tum rtabmap_gt.txt rtabmap_slam.txt -va --plot --plot_mode xy --align_origin
# replace this line with the command(s) you used
```

**Console output:**
``
![image](https://user-images.githubusercontent.com/19165811/236217042-0a51f2bc-dca3-4bde-ac7c-35a5d2a163b2.png)
`

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

sagardhatrak, 2023-05-04:
[image]

MichaelGrupp, 2023-05-08:
You are passing also the flag to do a full alignment with `-va`, which is short for `-v -a`, which is itself the short form of `--verbose --align`. Just use `-v` if you don't want the full alignment.

sagardhatrak, 2023-05-09:
Thank you for guidance. 
The following command has aligned the origin:
evo_ape tum rtabmap_gt.txt rtabmap_slam.txt --plot --plot_mode xy --align_origin

But, not able to apply full alignment. Does --align_origin includes full alignment? 
How can we do full alignment with --align_origin?
--align_origin and --align both are not working simultaneously (not aligning origin).

MichaelGrupp, 2023-05-09:
They can't be done simultaneously. Either the origin is aligned or there is a global alignment, they are two different things.