MichaelGrupp/evo #547: Coordinate horizontal axis (time stamp) problem

YoungHaulk, 2023-05-05:
**Description:**
Thanks a lot for such a great tool！
The image drawn by evo_traj has a negative time stamp, what is going on?

**Command:**
```
evo_traj tum vins_result_loop.txt gt.tum --ref gt.tum -pva
```

**Console output:**
```

Loaded 2359 stamps and poses from: vins_result_loop.txt
Loaded 6839 stamps and poses from: gt.tum
--------------------------------------------------------------------------------
Found 2240 of max. 2359 possible matching timestamps between...
	reference
and:	vins_result_loop.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning vins_result_loop.txt to reference.
Aligning using Umeyama's method...
Rotation of alignment:
[[-0.78734845 -0.28007563  0.54921768]
 [ 0.58568201 -0.61797446  0.52448465]
 [ 0.19250713  0.73461909  0.65059649]]
Translation of alignment:
[-1.82571805  0.26081025  0.63174804]
Scale correction: 1.0
--------------------------------------------------------------------------------
name:	vins_result_loop
infos:
	duration (s)	132.23000192642212
	nr. of poses	2240
	path length (m)	48.666036872304815
	pos_end (m)	[-1.68965359  0.52465932  0.79282515]
	pos_start (m)	[-1.85904959  0.30030635  0.62652387]
	t_end (s)	1641269993.156677
	t_start (s)	1641269860.926675
--------------------------------------------------------------------------------
name:	gt
infos:
	duration (s)	136.75339269638062
	nr. of poses	6839
	path length (m)	49.548782358657306
	pos_end (m)	[-1.85181844  0.35064384  0.87086713]
	pos_start (m)	[-1.75949538  0.29680422  0.51409811]
	t_end (s)	1641269993.2416167
	t_start (s)	1641269856.488224

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[image]

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu20.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```
[gt.txt](https://github.com/MichaelGrupp/evo/files/11404010/gt.txt)
[vins_result_loop.txt](https://github.com/MichaelGrupp/evo/files/11404012/vins_result_loop.txt)

MichaelGrupp, 2023-05-05:
I don't know yet why it's shown like this, might be a display issue of matplotlib? But as a workaround you could add `--plot_relative_time` to the command. Then the axis is showing values starting at zero, relative to the start time of your reference `gt.txt`.

YoungHaulk, 2023-05-06:
> I don't know yet why it's shown like this, might be a display issue of matplotlib with the large values of the UNIX timestamps? But as a workaround you could add `--plot_relative_time` to the command. Then the axis is showing values starting at zero, relative to the start time of your reference `gt.txt`.

Thank you very much for your quick answer！
After adding --plot_relative_time, the problem was solved！

Also, I have another problem. 
**Description:**
The yaw, pitch and roll in the resulting graph seem to have an overall offset from the groundtruth(please see attached image).
May I ask this kind of problem, is it because of the accuracy of my algorithm, or did I not set the relevant parameters when using evo_traj?

**The command used is:**
```
evo_traj tum vins_result_loop.txt gt.tum --ref gt.tum -pva --plot_relative_time
```
The two files above are still used（[gt.txt](https://github.com/MichaelGrupp/evo/files/11404010/gt.txt)
[vins_result_loop.txt](https://github.com/MichaelGrupp/evo/files/11404012/vins_result_loop.txt)）

[image]

closed 2023-06-23 by MichaelGrupp