MichaelGrupp/evo #551: The rpy angles of the estimated trajectory and the groundtruth do not align

YoungHaulk, 2023-05-30:
**Description:**
First of all, thank you very much for open source such an excellent evaluation software!
The xyz of the estimated trajectory and the groudtruth can be aligned.

[image]

But the rpy (roll, pitch, yaw) of the estimated trajectory is very different from the groudtruth.

[image]

Below is the output image of my reference algorithm whose rpy and ground truth are able to align. 
[image]

May I ask, what causes the estimated rpy and the true value to be misaligned?

**Command:**
```
evo_traj tum gt.tum ours.txt --ref gt.tum -pvas
```

**Console output:**
```

--------------------------------------------------------------------------------
Loaded 714 stamps and poses from: ours.txt
Loaded 5124 stamps and poses from: gt.tum
--------------------------------------------------------------------------------
Found 714 of max. 714 possible matching timestamps between...
	reference
and:	ours.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning ours.txt to reference.
Aligning using Umeyama's method... (with scale correction)
Rotation of alignment:
[[-1.14513318e-01  9.93421545e-01  5.77514589e-04]
 [ 3.12533992e-02  3.02157793e-03  9.99506926e-01]
 [ 9.92929970e-01  1.14474904e-01 -3.13938110e-02]]
Translation of alignment:
[1.47937297 0.88943569 0.15232245]
Scale correction: 1.6043280506217794
--------------------------------------------------------------------------------
name:	ours
infos:
	duration (s)	48.79999899864197
	nr. of poses	714
	path length (m)	40.92239120862965
	pos_end (m)	[1.60368235 0.91844864 0.19116145]
	pos_start (m)	[1.44418108 0.87139235 0.17972491]
	t_end (s)	1683433734.0103812
	t_start (s)	1683433685.2103822
--------------------------------------------------------------------------------
name:	gt
infos:
	duration (s)	51.24000000953674
	nr. of poses	5124
	path length (m)	40.468331729010316
	pos_end (m)	[1.621915 0.940646 0.228643]
	pos_start (m)	[1.423948 0.830845 0.182515]
	t_end (s)	1683433734.354

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.22.0
* Python version shown by `evo pkg --pyversion`: 3.8.10
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 20.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
![evo_config](https://github.com/MichaelGrupp/evo/assets/84961677/214aa530-bb5e-4876-9b5c-3b7bbe54cfaa)

```
[gt.txt](https://github.com/MichaelGrupp/evo/files/11597425/gt.txt)
[ours.txt](https://github.com/MichaelGrupp/evo/files/11597427/ours.txt)

YoungHaulk, 2023-05-30:
Sorry, the output of "evo_config show --brief --no_color" are:
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        8,
        8
    ],
    "plot_fontfamily": "serif",
    "plot_fontscale": 1.3,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 2,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "white",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

MichaelGrupp, 2023-05-31:
The poses look flipped. Probably because the poses in both trajectories are defined in different frames.

You can see it in the absolute angular error (with alignment) and in the plot with axis markers enabled:

```
evo_config set plot_axis_marker_scale 0.01 plot_pose_correspondences true

evo_ape tum gt.txt ours.txt --pose_relation angle_deg -asp --plot_mode xz
```

YoungHaulk, 2023-06-01:
> 

Thanks Reply！

The world coordinate system of the trajectory output by the algorithm is the IMU coordinate system in the first frame, and the poses output after that are all its IMU coordinate system relative to the world coordinate system. \
The trajectory of the groundtruth is the rigid body coordinate system relative to the world coordinate system of the mocap system
Now, the coordinate system of the rigid body is not consistent with the IMU, will this affect the rpy alignment?

In addition, does the command "evo_traj tum gt.txt ours.txt --ref gt.txt -pva" only align the position (xyz) of the trajectory, or align both the position (xyz) and the attitude (rpy) of the trajectory?

YoungHaulk, 2023-06-01:
I tried other trajectory comparisons, and the xyz positions can be aligned, but the rpy angles are very different. This is the command I use: "evo_traj tum gt2.tum ours2.txt --ref gt2.tum -pvas"
xyz:
[image]

rpy:

[image]

[gt2.txt](https://github.com/MichaelGrupp/evo/files/11620215/gt2.txt)
[ours2.txt](https://github.com/MichaelGrupp/evo/files/11620216/ours2.txt)

YoungHaulk, 2023-06-01:
@MichaelGrupp

MichaelGrupp, 2023-08-17:
The poses in your data still have different orientations. The alignment in evo can't fix this, since it's a rigid alignment of the whole trajectory (up to scale). So this means that you can have different global frames, but you need to ensure that the trajectories describe the motion of the same body frame.

closed 2023-08-17 by MichaelGrupp