MichaelGrupp/evo #554: How to obtain only the RPE value for the yaw, how to separately obtain the APE error for a specific axis.

NatureLan-sudo, 2023-06-05:
As indicated in the question, we know that currently we can calculate the overall RPE or APE errors using evo. However, for certain applications such as vehicle localization, we place greater emphasis on the accuracy of the yaw angle and the x and y directions. How can we achieve this functionality based on evo?

hiyyg, 2023-07-05:
Have you figured it out?

MichaelGrupp, 2023-07-14:
This is not available in the command line apps, and is also not planned from my side because metrics involving Euler angles for 3D input data are more something for custom metrics and too ~~fugly~~ error-prone for a general purpose implementation.

But it should be straight-forward to write a Python script that implements a custom metric for your specific needs.

As a start, you could load your trajectories and sync them. For example:

```
from evo.core import sync
from evo.tools import file_interface

traj_1 = file_interface.read_tum_trajectory_file("traj_1.txt")
traj_2 = file_interface.read_tum_trajectory_file("traj_2.txt")
traj_1, traj_2 = sync.associate_trajectories(traj_1, traj_2)
```

Euler angles can be accessed with e.g. `traj_1.get_orientations_euler(axes="sxyz")`, depending on the axis convention you use.
https://github.com/MichaelGrupp/evo/blob/9b7ec3a15e57bb7f2817b17f2ea676a0539e864d/evo/core/trajectory.py#L114

If you only want to get yaw values, you would use `get_orientations_euler(axes="sxyz")[:, 2]` in the example above. Doing this for both trajectories should allow you to calculate angle differences (reminder: don't just subtract angles...).

Similarly, positions are available via `traj_1.positions_xyz`. For example, to get only x & y, use `traj_1.positions_xyz[:, :2]`.

The rest is up to you.

---

**Alternative**: if you want to consider only x, y and yaw, why not save your trajectory data projected to 2D? evo's metrics can handle 2D data fine and the rotation angle is "yaw" in that dimension. Then you could just use the existing command line tools as usual.

closed 2023-07-20 by MichaelGrupp