MichaelGrupp/evo #555: Ros Map masking - wiki typo

JoLichtenfeld, 2023-06-08:
**Description:**
There seems to be a little error in the documentation of plotting, subsection 2D Ros maps / Masking.
The proposed command to disable masking is
`evo_config set ros_map_unknown_cell_value false`
This sets the corresponding value to the string "false" which cannot be parsed by [tools/plot.py ](https://github.com/MichaelGrupp/evo/blob/8681e8acc0b251f36f8153e39208ee26efed3d39/evo/tools/plot.py#L662).
So the command should either be
`evo_config set ros_map_unknown_cell_value 0`
or simply
`evo_config set ros_map_unknown_cell_value`.
The latter toggles the boolean state, so one would need to know the previous state.

MichaelGrupp, 2023-06-09:
Thanks for the info.

The problem seems to be that the "false" gets saved as a string when setting it. The code could handle a boolean ([here](https://github.com/MichaelGrupp/evo/blob/8681e8acc0b251f36f8153e39208ee26efed3d39/evo/tools/plot.py#L697)).

I will have a look how to improve this. Quick fix would be manually changing the json file and setting the value to `false` instead of `"false"`. Value 0 doesn't truly deactivate it and could hide meaningful data depending on the map image representation.

MichaelGrupp, 2023-06-12:
Added in v1.22.1, documentation is fixed. https://github.com/MichaelGrupp/evo/wiki/Plotting#masking

closed 2023-06-12 by MichaelGrupp