MichaelGrupp/evo #558: different time scale for the same dataset

Delbina, 2023-06-19:
hi
I want to know why the time scale is different in the following two plots? they are gained from the same dataset. one of them is from 0 to 200s, while the other one is from 0 to 600s?

[image]
[image]

MichaelGrupp, 2023-06-22:
One is absolute (see the exponent) and the other is relative to the start of your data. If you want to use relative in `evo_traj`, add the `--plot_relative_time` option.

Delbina, 2023-06-22:
thanks for your response
but what does it mean absolute and relative for mentioning the time?  
(what does the value of 450 mean in the time scale?)
would you please explain a bit more?

thanks

Delbina, 2023-06-22:
Thanks a lot

closed 2023-06-26 by MichaelGrupp