MichaelGrupp/evo #559: [ERROR] matrix is not a valid SO(3) group element

OsamaAlAhmadieh, 2023-06-21:
**Description:**
Hello, I had run my odometry algorithm on the Kitti sequence 00 data and got my estimated poses. Whenever I try to run the evo_rpe metric with the `pose_relation` set to `angle_deg` I get the following `[ERROR] matrix is not a valid SO(3) group element`. But running it with other arguments works. For example: with `pose_relation rot_part` works fune. 

**Command:**
```
evo_rpe kitti /media/datasets/dataset/poses/00.txt estimated_poses.txt --pose_relation angle_deg
```

**Console output:**
```

[ERROR] matrix is not a valid SO(3) group element
```

**Additional files:**
[estimated_poses.txt](https://github.com/MichaelGrupp/evo/files/11817562/estimated_poses.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.22.0
* Python version shown by `evo pkg --pyversion`: 3.10.6
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 22.04 (docker)
* did you change the source code? (yes / no):  no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

OsamaAlAhmadieh, 2023-06-21:
Was checking other issues and ran `evo_traj kitti estimated_poses.txt --full_check` and got the following: 

```--------------------------------------------------------------------------------
name:   00
infos:
        nr. of poses    4541
        path length (m) 3724.186990597451
        pos_end (m)     [-5.583931 -3.562758 96.96153 ]
        pos_start (m)   [ 5.551115e-17  3.330669e-16 -4.440892e-16]
checks:
        SE(3) conform   yes
        array shapes    ok
        quaternions     ok
stats:```

Seems my matrices aren't conforming. I wonder why it works for other metrics though. Shouldn't this check be run before calculating any type of metric?

OsamaAlAhmadieh, 2023-06-21:
seems the debugging is on my side, closing issue

OsamaAlAhmadieh, 2023-07-11:
I am reopening this issue since I am seeing some strange behavior still
Here is the file.
[estimated_poses.txt](https://github.com/MichaelGrupp/evo/files/12015203/estimated_poses.txt)
I have ran the [is_so3() check on it](https://github.com/MichaelGrupp/evo/blob/b329867d39c1f629ea7870f5c5aad7cf1879ea57/evo/core/lie_algebra.py#L176) and got true for all entries. 
But when I run the following command, I get the following: 
`evo_rpe kitti estimated_poses.txt gt_trunc.txt --pose_relation angle_deg --delta 1 --delta_unit d --plot --plot_mode xz`
I still get the following error: `[ERROR] matrix is not a valid SO(3) group element`
Not sure where the problem is. 
I have also ran the full check on the poses and got the following output: 
```--------------------------------------------------------------------------------
name:   estimated_poses
infos:
        nr. of poses    201
        path length (m) 145.37588367965617
        pos_end (m)     [47.9436  -9.77314 84.0541 ]
        pos_start (m)   [0. 0. 0.]
checks:
        SE(3) conform   yes
        array shapes    ok
        quaternions     ok
stats:```

gt_trunc is just the ground truth data from the kitti dataset sequence 0, truncated for the first 200 frames. here it is also for reference: 
[gt_trunc.txt](https://github.com/MichaelGrupp/evo/files/12015283/gt_trunc.txt)

**All other settings and configs remain the same as previously mentioned in the description**

MichaelGrupp, 2023-07-14:
Seems to be a float precision issue in the data. The poses pass the check, but in your command there are matrix multiplications happening that seem to result in less precise values that already don't pass the check anymore.

If you're blocked by this and the precision is not too far off, you can try to adjust the float tolerance here: https://github.com/MichaelGrupp/evo/blob/9b7ec3a15e57bb7f2817b17f2ea676a0539e864d/evo/core/lie_algebra.py#L184

A value of 1e-5 seems to make it pass. With such a small adaptation the result should be still fine to use.

OsamaAlAhmadieh, 2023-07-17:
Thank you @MichaelGrupp. Just as validation, my command will plot the error between 2 consecutive frames with the error being the angle difference between the estimated rotation and the ground truth rotation, right?

MichaelGrupp, 2023-07-18:
If you mean this command: `evo_rpe kitti estimated_poses.txt gt_trunc.txt --pose_relation angle_deg --delta 1 --delta_unit d --plot --plot_mode xz`

It means "relative angular error for every **1** **d**egree of motion".

> error between 2 consecutive frames with the error being the angle difference between the estimated rotation and the ground truth rotation

For this it would be `--pose_relation angle_deg --delta 1 --delta_unit f`, calculating the relative angular error for every motion between 2 consecutive poses (distance = **1** "**f**rame").

closed 2023-08-17 by MichaelGrupp