MichaelGrupp/evo #560: same set of trajectory data, maps drawn by evo_traj and evo_ape are different

2500035435, 2023-06-21:
**Description:**
**First of all, I would like to thank you for opening up such a great project, it helps a lot in slam work, but recently when using it, I met something that confused me :**
**For the same set of trajectory data, the trajectory maps drawn by evo_traj and evo_ape are different.
There are trajectory graphs and trajectory data below, hoping to find out what the reason is.**

[image]
The above trajectory, plotted by evo_traj, is the same as the one traveled in the real world and also the one displayed by the SLAM algorithm in RVIZ.

[image]
The evo_ape command is used to plot the trajectory above, and you can see that one of the right-angled turns in the real world trajectory is replaced with a straight line.

**Command:**
```
 `evo_traj tum ground_truth.txt vio.csv -p --plot_mode xyz`
`evo_ape tum -vas ground_truth.txt vio.csv -p --plot_mode xyz`

```
**Additional files:**

[ground_truth.txt](https://github.com/MichaelGrupp/evo/files/11817909/ground_truth.txt)
[vio.csv](https://github.com/MichaelGrupp/evo/files/11817915/vio.csv)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.12.0
* Python version shown by `evo pkg --pyversion`: 2.7.17
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2023-06-22:
`evo_ape` is only using poses that have matching timestamps and not all of the poses can be matched (read the print in `v`/`--verbose` mode), hence you get these gaps in your plot.

If you run `evo_traj` in a mode that requires the same time matching, you also get the same picture. In your case:

```
evo_traj tum vio.csv --ref ground_truth.txt -apv --plot_mode xyz
```

MichaelGrupp, 2023-06-22:
You need to check yourself why not all of your data has matching timestamps with the ground truth.

The only thing you could do in evo is playing around with the `--t_max_diff` parameter, but this is probably not going to improve things if there are time issues in your software/data.

2500035435, 2023-06-25:
Thanks alot, the issue was with the data timestamps. I set the parameter `--t_max_diff` to a value greater than 0.01, and now I can correctly plot the trajectory on evo_ape.

closed 2023-07-14 by MichaelGrupp