MichaelGrupp/evo #562: why umeyama alignment instead of hand eye calibration?

tanjunyao7, 2023-06-29:
Hi, 
I can't understand why Umeyama alignment is used to align the trajectories. It simply estimates the optimal transformation between two trajectories. But what if the two trajectories don't represent the motion of the same coordinate system?

Take the EuRoC dataset as an example. A VSLAM method estimates the trajectory of the camera frame. Some tracking system outputs the ground truth trajectory of the body frame. Even though the camera is rigidly mounted on the body, they represent different motions. In an extreme case where the UAV purely rotates around it's body frame Z axis, the trajectory of the camera is a circle while that of the body frame is a simple point. It's impossible to align a circle to a point. Even it could, it makes no sense. 

I think the correct way of dealing with trajectories of different frames is using hand eye calibration. That is, computing the most possible transformations between the two object frames and their corresponding world frames so that one really transform the motion of one into another and make the accuracy evaluation.

MichaelGrupp, 2023-07-14:
Calibration is different for every sensor setup and should be done before trajectory evaluation, this is not something that can be done in a meaningful way in this package here.

> But what if the two trajectories don't represent the motion of the same coordinate system?

Comparison of trajectories that don't describe the motion of the same moving frame generally doesn't make sense. Different inertial fixed frames are ok if you use alignment, but this doesn't change the shape of the trajectory.

Umeyama alignment just provides a way to measure similarity of trajectories.

> Take the EuRoC dataset as an example.

The EuRoC dataset provides precise extrinsic calibration transformation. So it is possible to accurately transform estimated poses from the camera frame by applying this static transformation - either in SLAM directly or by transforming each trajectory pose afterwards.

tanjunyao7, 2023-07-14:
no I mean one can not do such alignment even if the SLAM trajectory has zero drift.  Say we have two trajectories. One is camera to world transforms A from SLAM and the other is body to tracking system B as the ground truth. They can be transformed to each other by AX = YB with X be the transformation from body to camera and Y tracking system to world.  The alignment tries to calculate a transformation X that best transforms A to B: XA=B and compare XA and B either by RPE or APE. But such a transformation never exists even though the SLAM trajectory has no error. The equation just does not hold. 

So I think umeyama alignment can only be applied when the input trajectories describes the same motion. Otherwise it makes no sense. The better way is to do hand eye calibration from the input trajectories and compare Y^(-1)AX with B as the accuracy metric.

MichaelGrupp, 2023-07-14:
They're just two separate things. Differently worded:

Doing hand-eye, extrinsic or whatever calibration transformation if you have data of different moving frames should be done before evaluation and doesn't fit into this repository.

The purpose of Umeyama (or Horn) alignment in trajectory evaluation is to allow better understanding of how close shapes of trajectories match regardless of the inertial frame, not to compensate extrinsic offsets on the moving body. And it can be universally applied to trajectories of the same semantic moving frame, and has been used for that in robotics since 10+ years (I recommend the papers of Sturm & Kümmerle in that regard). You could also choose this alignment if you previously estimated hand-eye-calibration for your trajectories, it's just a property of the metric.

> So I think umeyama alignment can only be applied when the input trajectories describes the same motion. Otherwise it makes no sense.

If you re-read my comment, that's what I said. *Comparison of trajectories that don't describe the motion of the same moving frame generally doesn't make sense. Different inertial fixed frames are ok if you use alignment, but this doesn't change the shape of the trajectory (up to scale).*

closed 2023-07-14 by MichaelGrupp

gereon-t, 2023-07-18:
Hi! If I understand your question correctly, you want to account for a lever arm between the two frames, e.g. due to different mounting positions? If that is the case, you might want to have a look at this: https://github.com/gereon-t/trajectopy
This tool allows you to estimate not only the similarity transformation but also a lever arm. However, this requires the trajectory you want to align to have orientations available.
Also, if you have precise knowledge of the calibration of the system, you should use that instead of estimating the calibration, as MichaelGrupp has already pointed out.

tanjunyao7, 2023-07-20:
I have thought that the ground truth supplied by the datasets(TUM, EuRoc, Kitti etc) denotes the trajectory of the body frame. But after some investigation, I found that the publishers already converted the body ground truth path to the cameras. Then it's justified to align them. 
I have this question because my colleagues always directly compare a camera trajectory with a lidar trajectory using evo.