MichaelGrupp/evo #575: Unusually High Relative Pose Error

zhimou, 2023-08-03:
**Description:**
I tried to use evo to benchmark slam algorithm using RTK as reference,  and I recorded RTK and fast lio trajectories in TUM format. I found the fast lio trajectory align RTK reference very well but the relative pose error is abnormally large in more than half of the trajectory. 
  
[image]

As shown above, the trajectory initiates in the top right corner and close the loop in the end. According to color we can see the relative pose error increase then decrease and the relative pose error should not as high as the color indicated (more than 5 meters in more than half of the trajectory!)

[image]

Let us zoom in and look closer at the button right corner in above graph, there is no way that the relative pose error is as high as 10 meter unless there is a serious problem with time sync. I have already verified the timestamp in rviz and found no noticeable time offset that could account for this.
**Command:**
```
evo_rpe tum utm_positions_seq4.txt  odom_positions_fast_lio_seq4.txt --delta_unit m --delta 5   -a --pairs_from_reference  --t_max_diff 0.1 -p --plot_mode=xy --verbose
```

**Console output:**
```
--------------------------------------------------------------------------------
Loaded 5157 stamps and poses from: utm_positions_seq4.txt
Loaded 5156 stamps and poses from: odom_positions_fast_lio_seq4.txt
--------------------------------------------------------------------------------
Synchronizing trajectories...
Found 5156 of max. 5156 possible matching timestamps between...
	utm_positions_seq4.txt
and:	odom_positions_fast_lio_seq4.txt
..with max. time diff.: 0.1 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning using Umeyama's method...
Rotation of alignment:
[[-0.07981264  0.99680988  0.        ]
 [-0.99680988 -0.07981264  0.        ]
 [ 0.          0.          1.        ]]
Translation of alignment:
[-0.11044243 -0.21151651  0.        ]
Scale correction: 1.0
--------------------------------------------------------------------------------
Found 125 pairs with delta 5.0 (m) among 5156 poses using consecutive pairs.
Compared 125 relative pose pairs, delta = 5.0 (m) with consecutive pairs.
Calculating RPE for translation part pose relation...
--------------------------------------------------------------------------------
RPE w.r.t. translation part (m)
for delta = 5.0 (m) using consecutive pairs
(with SE(3) Umeyama alignment)

       max	10.221452
      mean	5.869984
    median	7.119585
       min	0.453892
      rmse	7.029179
       sse	6176.170400
       std	3.866865

--------------------------------------------------------------------------------

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[odom_positions_fast_lio_seq4.txt](https://github.com/MichaelGrupp/evo/files/12249033/odom_positions_fast_lio_seq4.txt)
[utm_positions_seq4.txt](https://github.com/MichaelGrupp/evo/files/12249034/utm_positions_seq4.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.24.0
* Python version shown by `evo pkg --pyversion`: 3.10.9
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2023-08-03:
Looks exactly like https://github.com/MichaelGrupp/evo/issues/482

Short answer: use `--pose_relation point_distance` when your data doesn't have orientation.

zhimou, 2023-08-03:
Thanks Michael!!! It solved my problem.

> Looks exactly like #482
> 
> Short answer: use `--pose_relation point_distance` when your data doesn't have orientation.

closed 2023-08-03 by zhimou