MichaelGrupp/evo #579: The unit error of `rotation_angle_(deg/rad)` in metrics.py

KuangHaofei, 2023-08-10:
Hi,

Thanks for your excellent work!

Now, I found whatever I choose `rotation_angle_rad` or `rotation_angle_deg` will link to the same unit `"rotation angle"`:
https://github.com/MichaelGrupp/evo/blob/5cf21800b07c9cb2bbbec19d4539e134a0bdbaa9/evo/core/metrics.py#L67C1-L68

So, in the `APE` class, choosing `rotation_angle_rad` or `rotation_angle_deg` will set the `unit` to `Unit.degrees`:
https://github.com/MichaelGrupp/evo/blob/5cf21800b07c9cb2bbbec19d4539e134a0bdbaa9/evo/core/metrics.py#L361C1-L364C37.
And the `APE` class will only compute the rotation error in rad:
https://github.com/MichaelGrupp/evo/blob/5cf21800b07c9cb2bbbec19d4539e134a0bdbaa9/evo/core/metrics.py#L426C1-L431C79

There are similar problems with the `RPE` class in the [metrics.py](https://github.com/MichaelGrupp/evo/blob/master/evo/core/metrics.py).
Could you please check the [metrics.py](https://github.com/MichaelGrupp/evo/blob/master/evo/core/metrics.py) to fix the error?

Thanks!

closed 2023-08-17 by MichaelGrupp

MichaelGrupp, 2023-08-17:
Thanks for reporting this. It was indeed an error that was recently introduced. Fixed it now.

KuangHaofei, 2023-08-17:
That's great! Thanks!

PR #581 (2023-08-17): Fix pose relation enum
Fixes a bug introduced recently in PR #569 and switches all enums to `@unique`
to catch this in the future.

Fixes: #579
commit 6c27ad70 (2023-08-17): Fix pose relation enum (#581)

Fixes a bug introduced recently in PR #569 and switches all enums to @unique
to catch this in the future.

Fixes: #579