MichaelGrupp/evo #590: How to automatically close this window

liutaosgsg, 2023-09-15:
How to automatically close this window
[image]

MichaelGrupp, 2023-09-15:
?

If you don't want to see the window, don't put `--plot` or `-p` in your command. That said, it's a GUI window that is designed to be closed by the user with a mouse click, or e.g. alt-F4 on Linux.

If you want to do something beyond that, please research yourself how to do that with your OS.

liutaosgsg, 2023-09-18:
> ?
> 
> If you don't want to see the window, don't put `--plot` or `-p` in your command. That said, it's a GUI window that is designed to be closed by the user with a mouse click, or e.g. alt-F4 on Linux.
> 
> If you want to do something beyond that, please research yourself how to do that with your OS.

Thanks，I need image，but I don't want to show this window，or how to close this window in command。

liutaosgsg, 2023-09-18:
[image]
[image]

MichaelGrupp, 2023-09-18:
Images can be saved directly without GUI window with `--save_plot`, for example: `--save_plot my_plot.png`, `--save_plot my_plots.pdf`