MichaelGrupp/evo #591: Importing apply_settings from evo.tools.plot doesn't work

varunagrawal, 2023-09-20:
**Description:**

I ran `pip install -U evo` and got the latest package (version 1.25.1).

When I try to change the plot settings, I get an error that the `apply_settings` method doesn't exist.

Cloning the repo and installing it with `pip install -e .` doesn't raise the same error, which makes me guess that the issue is in the packaging of v1.25.1. Perhaps the uploaded package didn't use the correct commit/release version?

You should be able to reproduce this by executing the interactive metrics notebook in this repo.

**Command:**
```
from evo.tools import plot
plot.apply_settings  # AttributError!
```

```
from evo.tools.plot import apply_settings. # throws ImportError
```

**Console output:**
Case 1:
```
AttributeError: module 'evo.tools.plot' has no attribute 'apply_settings'
```

Case 2:
```
ImportError: cannot import name 'apply_settings' from 'evo.tools.plot'
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.25.1
* Python version shown by `evo pkg --pyversion`: 3.11.5
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): macOS, can also reproduce on Windows 11 and Ubuntu 20.04.
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```json
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```

Bastiaan-Sw, 2023-09-25:
Any solution yet? I am facing same problem. Tried most of the different versions without any success.

MichaelGrupp, 2023-10-09:
I guess the issue is that you use the notebook from the master branch that is ahead of v1.25.1. The change that introduced `apply_settings` was not released yet (I plan to do that soon with some other things).
In that case, I would recommend to check out the 1.25.1 of the repository to get the matching version of the notebook.

varunagrawal, 2023-10-09:
That's what we're currently doing: installing via the GitHub repository. I wanted to bring this to your attention to both track the issue as well as because non-expert users are having trouble with this.

Looking forward to the new release.

MichaelGrupp, 2023-11-14:
New version tag that includes this change was released today.

closed 2023-11-14 by MichaelGrupp