MichaelGrupp/evo #593: Question about Umeyama alignment

robincourant, 2023-09-26:
I want to extend my appreciation for this project - it's been a valuable tool for my work! Thanks!

I'm using Umeyama alignment to work with trajectories consisting of 9 poses. They all share the same rotations initially but have different translations. 
When I align them, it works fine for the translations, but sometimes it flips the rotations, even though they were identical before alignment.
Has anyone else experienced this behaviour? Any ideas why it's happening?

MichaelGrupp, 2023-10-09:
This sounds like a question related to your specific data, so this can't really be answered without a reproducible example with trajectory files.

robincourant, 2023-10-09:
Hi, thanks for your feedback, I've attached two trajectory files ([traj_1.txt](https://github.com/MichaelGrupp/evo/files/12845515/traj_1.txt) and [traj_2.txt](https://github.com/MichaelGrupp/evo/files/12845516/traj_2.txt)) that produce the error I described. Am I doing something wrong? I've found an alternative with a leat square on only scale an translation.

MichaelGrupp, 2023-10-10:
Looking at the two trajectories, this seems to be the expected result because the translations go into flipped directions relative to the first pose of each trajectory.

You can see this with origin alignment `--align_origin`, which forces the two trajectories to start from the same position and orientation. I also added `--correct_scale` because the shape of the trajectory seems to be the almost same apart from scale.

```
evo_traj kitti --ref=traj_2.txt traj_1.txt --correct_scale --align_origin --plot_mode xyz -p
```

[image]
[image]

Umeyama alignment minimizes the distance between corresponding positions. So in your case you get well-aligned positions, but the orientations don't match because the translations are going into opposite directions relative to the poses. But this is expected with this kind of data.

[image]

closed 2023-10-11 by robincourant