MichaelGrupp/evo #597: can not use parameter -p. No module named 'tkinter'

WangPei-hitman, 2023-10-08:
**Description:**
With Ubuntu 22.04 no ROS1 , I have to change another way to use evo. I find the Dockerfile.rosnoetic file in sources and build it. But in container , the evo_traj can not use -p parameter.
can not use parameter -p. No module named 'tkinter'
**Command:**
```
docker run -it \
-p 127.0.0.1:8111:80 \ 
-v /home/wp/bag:/bag \
--name evo_container \
-v /tmp/.X11-unix:/tmp/.X11-unix \
-e DISPLAY=unix$DISPLAY \
-e GDK_SCALE \
-e GDK_DPI_SCALE \
evo:v1.25.1 /bin/bash

 evo_traj bag 2023-10-08-17-18-45.bag --all_topics -p
```

**Console output:**
```
--------------------------------------------------------------------------------
name:   /lviorf/mapping/odometry
infos:  7839 poses, 56761.287m path length, 783.771s duration
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/evo/entry_points.py", line 95, in launch
    main_module.run(args)
  File "/usr/local/lib/python3.8/dist-packages/evo/main_traj.py", line 279, in run
    fig_xyz, axarr_xyz = plt.subplots(3, sharex="col",
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 1501, in subplots
    fig = figure(**fig_kw)
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/_api/deprecation.py", line 454, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 840, in figure
    manager = new_figure_manager(
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 383, in new_figure_manager
    _warn_if_gui_out_of_main_thread()
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 361, in _warn_if_gui_out_of_main_thread
    if _get_required_interactive_framework(_get_backend_mod()):
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 208, in _get_backend_mod
    switch_backend(rcParams._get("backend"))
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/pyplot.py", line 271, in switch_backend
    backend_mod = importlib.import_module(
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/backends/backend_tkagg.py", line 1, in <module>
    from . import _backend_tk
  File "/usr/local/lib/python3.8/dist-packages/matplotlib/backends/_backend_tk.py", line 8, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.25.1
* Python version shown by `evo pkg --pyversion`: 3.8.10
* operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 22.04 but use dockerfile.noetic
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```
Thanks for your reply!

WangPei-hitman, 2023-10-09:
if I use Agg as plot_backend ,that means the output of `evo_config show --brief --no_color`:
```
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```
Then the out put of my command `evo_traj bag 2023-10-08-17-12-46.bag --all_topics -p` is:
```
--------------------------------------------------------------------------------
name:	/lviorf/mapping/odometry
infos:	3314 poses, 11525.862m path length, 331.288s duration
/usr/local/lib/python3.8/dist-packages/evo/tools/plot.py:193: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.show()

```
What GUI backend should I use?

MichaelGrupp, 2023-10-09:
The Docker files in this repository are targeted towards CI rather than interactive usage with GUI forwarding. I assume that just some dependencies for the GUI are missing from the Docker image, as can be seen in your error. Probably you need to adapt the Docker build for your purposes, i.e. install also tkinter etc.

MichaelGrupp, 2023-11-14:
Closing because of inactivity

closed 2023-11-14 by MichaelGrupp