MichaelGrupp/evo #599: Questions about Umeyama Alignment and Evaluation Results

wangjinhoon, 2023-10-30:
Hello, I wonder why the results of the two statements are different. Can't evo_ape also get the right traj like evo_traj??
```
evo_traj tum 1030test31.14.txt --ref 1030test32.5.txt --plot
```
[image]

```
evo_ape tum 1030test32.5.txt 1030test31.14.txt --plot
```
[image]

thank!

MichaelGrupp, 2023-10-30:
`evo_ape` by default shows only the synchronized poses that were used for the metrics calculation. But you can show the "full" reference with `--plot_full_ref`, i.e. including poses that were not used by the metric.

`evo_traj` shows the full reference because it can align multiple trajectories.

wangjinhoon, 2023-10-30:
Thank you for your quick reply.

 Does synchronization mean that Roll, Pitch, and Yaw are the same on the same timestamp when aligned?

I want to compare the Vive tracker with the VO. Is there anything I should care about when creating test data to increase the number of synchronized poses? Currently, my data is very different in the amount of data and does not take into account roll, pitch, or yaw.

MichaelGrupp, 2023-11-14:
Umeyama alignment rigidly transforms (optionally with scale) your trajectory so that it fits the best way possible to the other one. This doesn't mean that the RPY angles are the same afterwards.

wangjinhoon, 2023-11-16:
I increased the data volume and resolved the issue. Thank you

closed 2023-11-16 by MichaelGrupp