MichaelGrupp/evo #602: plot error with matplotlib 3.8.1

sacrover, 2023-11-14:
Referring to this [matplotlib 3.8 issue](https://github.com/matplotlib/matplotlib/issues/26831#issue-1903676455), the color cycle iterator, 'prop_cycler'  was removed. This breaks the `--plot` option. 

**Matplotlib 3.7 works well**. You can change this in  [setup.py](https://github.com/MichaelGrupp/evo/blob/13d55881734309815492e097bc59dccefaf47f82/setup.py#L96C11-L96C11) or downgrade to matplotlib version 3.7 after installing evo.

MichaelGrupp, 2023-11-14:
Thanks for the report. Replacing it with a call to the seaborn API seems to be enough:
```
color_palette = itertools.cycle(sns.color_palette())
next(color_palette)
```

I'm fixing some other things that showed up after the matplotlib upgrade and will include this in there.

closed 2023-11-14 by MichaelGrupp

MichaelGrupp, 2023-11-14:
Fixed in v1.25.2

commit 2b7aa51c (2023-11-14): Use seaborn color_palette instead of private matplotlib API

Fixes: #602
PR #595 (2023-11-14): Use correct matplotlib public interfaces for type hints.
https://github.com/matplotlib/matplotlib/issues/26812

Also fixes access to colors. Closes: #602
commit aa115ab2 (2023-11-14): Use correct matplotlib public interfaces for type hints. (#595)

matplotlib/matplotlib#26812

Also fixes access to colors. Closes: #602