MichaelGrupp/evo #608: Pandas ValueError: Multi-dimensional indexing (e.g. `obj[:, None]`) is no longer supported.

aserbremen, 2023-11-20:
First of thanks for this toolbox and all your work!

**Description:**

I get `ValueError: Multi-dimensional indexing (e.g. obj[:, None]) is no longer supported.¸`
when running any evo_res command due to new versions of `pandas` > 1.2.5. 

I installed evo with version 1.2.5. in a micromamba virtual environment and its working with `evo pkg --version` 1.25.2

**Command:**
```
evo_res atlas_rpe.zip atlas_scan_matching_odom_rpe.zip -p --use_filenames
```

**Command Line Output:**
```
[WARNING] duplicate indices in error array of atlas_scan_matching_odom_rpe.zip - keeping only first occurrence of duplicates

RPE w.r.t. translation part (m)
for delta = 1 (frames) using consecutive pairs
(with SE(3) Umeyama alignment)
                                      rmse      mean    median       std  
atlas_rpe.zip                     0.733409   0.67149  0.645417   0.29494   
atlas_scan_matching_odom_rpe.zip  0.246537  0.098591  0.004133  0.225966   

                                       min       max         sse  
atlas_rpe.zip                     0.112185  1.691492   35.500684  
atlas_scan_matching_odom_rpe.zip  0.000043  1.579923  200.272305  

[ERROR] Unhandled error in evo.main_res
Traceback (most recent call last):
  File "/home/andi/.local/lib/python3.10/site-packages/evo/entry_points.py", line 95, in launch
    main_module.run(args)
  File "/home/andi/.local/lib/python3.10/site-packages/evo/main_res.py", line 217, in run
    dist_grid.map(sns.distplot, metric_label)  # fits=stats.gamma
  File "/home/andi/.local/lib/python3.10/site-packages/seaborn/axisgrid.py", line 752, in map
    self._facet_plot(func, ax, plot_args, kwargs)
  File "/home/andi/.local/lib/python3.10/site-packages/seaborn/axisgrid.py", line 848, in _facet_plot
    func(*plot_args, **plot_kwargs)
  File "/home/andi/.local/lib/python3.10/site-packages/seaborn/distributions.py", line 2511, in distplot
    kdeplot(**{axis: a}, ax=ax, color=kde_color, **kde_kws)
  File "/home/andi/.local/lib/python3.10/site-packages/seaborn/distributions.py", line 1717, in kdeplot
    p.plot_univariate_density(
  File "/home/andi/.local/lib/python3.10/site-packages/seaborn/distributions.py", line 999, in plot_univariate_density
    artist, = ax.plot(support, density, **artist_kws)
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_axes.py", line 1632, in plot
    lines = [*self._get_lines(*args, data=data, **kwargs)]
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_base.py", line 312, in __call__
    yield from self._plot_args(this, kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_base.py", line 487, in _plot_args
    x = _check_1d(xy[0])
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 1327, in _check_1d
    ndim = x[:, None].ndim
  File "/home/andi/.local/lib/python3.10/site-packages/pandas/core/indexes/base.py", line 5199, in __getitem__
    disallow_ndim_indexing(result)
  File "/home/andi/.local/lib/python3.10/site-packages/pandas/core/indexers/utils.py", line 343, in disallow_ndim_indexing
    raise ValueError(
ValueError: Multi-dimensional indexing (e.g. obj[:, None]) is no longer supported. Convert to a numpy array before indexing instead.

[ERROR] evo module evo.main_res crashed - no logfile written (disabled)
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.25.2
* Python version shown by `evo pkg --pyversion`: 3.10.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 22.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2023-11-20:
This looks exactly like this dependency problem: https://github.com/mwaskom/seaborn/issues/3312

I tried downgrading matplotlib in my virtual environment and can reproduce it with:
* matplotlib 3.4.0
* pandas 2.0.3
* seaborn 0.13.0

The bug doesn't happen when using a more recent matplotlib version than that. So I would recommend `pip install --upgrade matplotlib` in your environment, too.

aserbremen, 2023-12-04:
Thanks for the quick response. Either upgrading matplotlib or downgrading pandas seems to be the solution here

closed 2023-12-04 by aserbremen