MichaelGrupp/evo #612: Data conversion

Minsu-Kim-KNU, 2023-12-07:
Hi, I wanna convert my rostopic which is consist of  x, y, z pose from the SLAM algorithm, but EVO requires KITTI format or the TUM format.

I wanna try to use KITTI format, which is consist of 4x4 homongeous matrix, with rotational matrix, and translation matrix T.
I have also IMU information with 6 and 9 axis, 

If in case of 6-axis IMU, how can I convert my data to the KITTI format with 12 elements?

plz. thank you

MichaelGrupp, 2023-12-11:
IMU messages can't be loaded by evo, only formats that represent poses ( / points). The processing of IMU data is expected to be done by some external algorithm that outputs a pose (SLAM, odometry, ...).

Minsu-Kim1822, 2023-12-11:
@MichaelGrupp 
Thank you for your advice. I recoreded all my SLAM trajectories and ground truth trajectories with orientation information, then I combine and modify as KITTI dataset format. thank you for your help.

closed 2023-12-11 by Minsu-Kim1822