MichaelGrupp/evo #614: Unable to convert recorded trajectory bag file to kitti format

boybestak99, 2023-12-13:
**Description:**
Hi, I am trying to convert the trajectory bag file which I have recorded using A-Loam slam algorithm. However, when I run the command below. I experience error in converting the bag file into kitti format. I have attached my bag file in .txt format.

**Command:**

evo_traj bag aloam_trajectory.bag /aft_mapped_path --save_as_kitti

**Console output:**

[ERROR] unsupported message type: nav_msgs/Path

Additional info (if needed)
evo version  v1.12.0
python version 2.7.17
ubuntu 18.04
* output of `evo_config show --brief --no_color`: 
{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
}

[Uploading output_data.txt…]()

MichaelGrupp, 2023-12-15:
As the error says, `nav_msgs/Path` topics are not supported. The supported topic types are listed here: https://github.com/MichaelGrupp/evo/wiki/Formats#bag--bag2---ros1ros2-bagfile

boybestak99, 2023-12-16:
Thank you for your reply. Is there a way to convert into the supported formats?

MichaelGrupp, 2024-01-05:
There is a way but you need to code your own script that loads the Path messages and converts them to a PoseStamped topic. Should be relatively easy in Python with the rosbag API.

closed 2024-01-20 by boybestak99