MichaelGrupp/evo #616: Rotations don't match after alignment

qinjunj, 2023-12-22:
Hi there. I'm applying `-a` to align two trajectories when plotting them. However, this alignment only works for the `xyz` part (shown in the first two pics), but not the rotation part (as in the last two pics).
[image]
[image]
[image]
[image]

From the last two pictures, I can see that the rotation part (roll, pitch, yaw) will be better aligned if I switch roll and pitch, and probably make some rotation around the yaw axis. I understand that this will not be taken care of by the Umeyama algorithm because it is based on minimizing the difference in the `xyz` part only #299. What I'm not sure is, does this misalignment imply that the rotation has a different coordinate system as the `xyz` part? If so, is it valid if I transform the rotation part manually in the way stated above?

Thanks for any guidance!

MichaelGrupp, 2024-01-05:
In this case, you would need to investigate the orientations in your data. For example you could enable pose axis markers to see them in plots:
https://github.com/MichaelGrupp/evo/wiki/Plotting#coordinate-axis-markers

...and/or use the RPE or APE with `--pose_relation angle_deg` to check the rotation errors between the two trajectories.

LeeBY68, 2024-01-14:
I have the same problem (Umeyama ONLY aligns positions instead of rotation) with evaluation using EVO in TextSLAM, inducing the wrong RPE results.

In response to this, I've developed a complementary solution for EVO that addresses the problem. You can use and find the details in the GitHub project: https://github.com/SJTU-ViSYS/SLAM_Evaluation/.

closed 2024-02-05 by MichaelGrupp