MichaelGrupp/evo #623: [ERROR] matrix is not a valid SO(3) group element

DeepDuke, 2024-01-11:
**Description:**
It cannot compute angle error in degree but computation for rotation vector is fine.

**Command:**
```
 evo_ape gt_poses.txt carla_est_poses_from_00_to_01_constant_vel_model.txt -a -r angle_deg
```

**Console output:**
```
[ERROR] matrix is not a valid SO(3) group element
```

**Additional files:**
Here are my psoes files.
[gt_poses.txt](https://github.com/MichaelGrupp/evo/files/13899009/gt_poses.txt)
[carla_est_poses_from_00_to_01_constant_vel_model.txt](https://github.com/MichaelGrupp/evo/files/13899019/carla_est_poses_from_00_to_01_constant_vel_model.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```
evo pkg --version
v1.25.2

evo pkg --pyversion
3.9.17

Did you change the source code? NO

OS: Ubuntu  20.04

evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

DeepDuke, 2024-01-11:
Well. I modified the atol of `is_so3` in `lie_algebra.py` to fix it as 
```
def is_so3(r: np.ndarray) -> bool:
    """
    :param r: a 3x3 matrix
    :return: True if r is in the SO(3) group
    """
    atol = 1e-2
    print('is_so3: atol = {}'.format(atol))
    # Check the determinant.
    # det_valid = np.allclose(np.linalg.det(r), [1.0], atol=1e-6)
    det_valid = np.allclose(np.linalg.det(r), [1.0], atol=atol)
    
    # Check if the transpose is the inverse.
    # inv_valid = np.allclose(r.transpose().dot(r), np.eye(3), atol=1e-6)
    inv_valid = np.allclose(r.transpose().dot(r), np.eye(3), atol=atol)
    return det_valid and inv_valid
```

why not pass `atol` out as  a user-specified argument?

MichaelGrupp, 2024-07-30:
Closing, I don't see a big benefit in making this configurable. 0.01 is quite imprecise, and it's better to restrict this by default to avoid follow-up errors in calculations.

closed 2024-07-30 by MichaelGrupp