MichaelGrupp/evo #625: RPE values is wrong

BamsaAhmed, 2024-01-24:
I used evo for find APE and RPE for my SLAM, the ape give correct value and curve, while the rpe curve is not correct for the same data?
Both curves is shown below:
[image]
[image]

could anyone tell me why?

BamsaAhmed, 2024-01-24:
note the color not changed along the path for rpe

LeeBY68, 2024-01-26:
seem your rpe visualization did not aligned.

MichaelGrupp, 2024-02-05:
Without specifying why you think it's "wrong", we can't help you. With the current information given, it looks like you might just be confused by the missing alignment as mentioned by @LeeBY68 . Feel free to re-open if you have specific questions.