MichaelGrupp/evo #626: error when i plot traj."evo module evo.main_traj crashed - no logfile written (disabled)"

Victoria-aaa, 2024-01-27:
**Description:**
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/victoria/.local/lib/python3.6/site-packages/evo/entry_points.py", line 92, in launch
    main_module.run(args)
  File "/home/victoria/.local/lib/python3.6/site-packages/evo/main_traj.py", line 398, in run
    from evo.tools import plot
  File "/home/victoria/.local/lib/python3.6/site-packages/evo/tools/plot.py", line 34, in <module>
    import matplotlib.pyplot as plt
  File "/home/victoria/.local/lib/python3.6/site-packages/matplotlib/pyplot.py", line 2336, in <module>
    switch_backend(rcParams["backend"])
  File "/home/victoria/.local/lib/python3.6/site-packages/matplotlib/pyplot.py", line 276, in switch_backend
    class backend_mod(matplotlib.backend_bases._Backend):
  File "/home/victoria/.local/lib/python3.6/site-packages/matplotlib/pyplot.py", line 277, in backend_mod
    locals().update(vars(importlib.import_module(backend_name)))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/victoria/.local/lib/python3.6/site-packages/matplotlib/backends/backend_tkagg.py", line 1, in <module>
    from . import _backend_tk
  File "/home/victoria/.local/lib/python3.6/site-packages/matplotlib/backends/_backend_tk.py", line 6, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

**Command:**
evo_traj kitti KITTI_00_gt.txt -p

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2024-02-05:
Tk / tkinter is not installed. Normally, this is part of Python distros. If you're missing it, please search in the internet how to install it on your specific operating system.

closed 2024-02-05 by MichaelGrupp

akshaygulabrao, 2025-05-20:
I also ran into this issue and installing tk / tkinter did not help.

However, using Qt5 as the backend completely resolved this issue for me.