MichaelGrupp/evo #633: Degenerate covariance rank, Umeyama alignment is not possible

joshuaoreilly, 2024-02-10:
When running the following command on two trajectories:

```
evo_traj tum /mnt/storage/trajectories_zed_tum.txt --ref=trajectories_spot_tum.txt --t_max_diff 10000000000000000000000000 --align --correct_scale -p --plot_mode=xy
```

I get the following error:

```
[ERROR] Degenerate covariance rank, Umeyama alignment is not possible
```

This also occurs without `--correct_scale`. I'm not well versed in the theory behind how Umeyama alignment works; I know that at the very least, I don't have the case of [static motion](https://github.com/MichaelGrupp/evo/pull/240), but I'm otherwise at a loss as to the cause.

[trajectories_spot_tum.txt](https://github.com/MichaelGrupp/evo/files/14228466/trajectories_spot_tum.txt)
[trajectories_zed_tum.txt](https://github.com/MichaelGrupp/evo/files/14228467/trajectories_zed_tum.txt)

evo version: v1.26.0
Python version: 3.10.12
OS: Ubuntu 22.03
Changed source code: No
Output of `evo_config show --brief --no_color`:

```
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```

MichaelGrupp, 2024-02-10:
> --t_max_diff 10000000000000000000000000

Hmm... :smile: 

The timestamps in the trajectories are completely different. Either fix the files or use `--t_offset` in evo if you know the exact time offset.

When you do what you did with `--t_max_diff`, all poses in the second trajectory are matched only to the first pose of the other trajectory because that's then the "best fitting" timestamp. Alignment against only one pose doesn't work.

joshuaoreilly, 2024-02-10:
> > --t_max_diff 10000000000000000000000000
> 
> Hmm... 😄
> 
> The timestamps in the trajectories are completely different. Either fix the files or use `--t_offset` in evo if you know the exact time offset.
> 
> When you do what you did with `--t_max_diff`, all poses in the second trajectory are matched only to the first pose of the other trajectory because that's then the "best fitting" timestamp. Alignment against only one pose doesn't work and causes the rank issue in the Umeyama algorithm.

Ah, okay, I think I'm trying to achieve the opposite of what evo does, then; we have two trajectories that *should* be following the same trajectory (both cameras were mounted on the same robot and recorded simultaneously), but whose timestamps have an unknown offset we're trying to find. I don't suppose you know any tools that would be suitable for this purpose?

MichaelGrupp, 2024-02-12:
Yes, time synchronization is not part of this package. It depends too much on the specific recording and SLAM setup that is used. 

You might find tools for specific systems, or papers describing how time sync was done (for example, the TUM RGBD benchmark paper by Sturm et al. describes how they synced the camera to their mocap system.)

joshuaoreilly, 2024-02-12:
Perfect, thank you so much!

closed 2024-02-12 by joshuaoreilly