MichaelGrupp/evo #636: Rosbags update introduces breaking change

CameronFraser, 2024-03-01:
Yesterday rosbags released an update that introduces a breaking change to this project: https://gitlab.com/ternaris/rosbags/-/commit/5aacc7eb800f1c0299d263bbdff68f3eeaff8e6b

Specifically around the serde imports

```
    from rosbags.serde.serdes import cdr_to_ros1
E   ModuleNotFoundError: No module named 'rosbags.serde.serdes'
```

Suggest pinning to an earlier version of rosbags

wanghao15, 2024-03-02:
Thank you for pointing out the issue with the latest update of rosbags and suggesting a workaround by pinning to an earlier version. it works.

HsuanPin31, 2024-03-03:
if i am using ros noetic, what version of rosbag should i  pin to?

Pinklet, 2024-03-03:
> @CameronFraser if i am using ros noetic, what version of rosbag should i pin to?

I have the same problem! Do u solve it?

Pinklet, 2024-03-03:
> @CameronFraser if i am using ros noetic, what version of rosbag should i pin to?

I Comment out the "from rosbags.serde.serdes import cdr_to_ros1" at evo/tools/file_interface.py and success！

HsuanPin31, 2024-03-03:
it works!!! you are a life safer @Pinklet

MichaelGrupp, 2024-03-04:
Thanks for the report, I'll release a patch with a hotfix to pin the version.

I'll then look into a proper fix here: https://github.com/MichaelGrupp/evo/issues/638

rosbags is a great package and IMP the only viable option for cross-platform & cross ROS version data compatibility, but unfortunately has not reached a major version that has API stability...

MichaelGrupp, 2024-03-05:
Hotfix is rolled out in `v1.26.1`. If you already got that rosbags version, you can do: `pip install rosbags==0.9.19` to downgrade for now, which has the same effect as the hotfix.

closed 2024-03-19 by MichaelGrupp