MichaelGrupp/evo #642: Extract the data from xyz_view

xbnz-1997, 2024-03-13:
How to save xyz_view data in evo_traj? I want to solve the error data of XYZ separately.

MichaelGrupp, 2024-03-19:
"xyz_view" data are just the X, Y, Z positions of the trajectory. So it's part of the input trajectory file, the trajectory object in the code and the output if you use one of the `--save_as_*` options.

xbnz-1997, 2024-03-20:
I mean xyz_view when the real trajectory is aligned with the contrast trajectory.

MichaelGrupp, 2024-03-20:
Here you would need one of the saving options: https://github.com/MichaelGrupp/evo/wiki/Formats#saving--exporting-to-other-formats
Export of only X Y Z is not available in the command line apps, but would be possible with a small code modification if you want to do that in a fork.