MichaelGrupp/evo #643: Distortion occurs when comparing with the true trajectory

HowieHepburn, 2024-03-16:
**Description:**
My trajctory :(evo_traj tum lio_odom_UrbanTST.tum -p):
[image]

GroundTruth:(evo_traj tum gt_slam_UrbanNav-HK-TST-20210517.csv -p):
[image]

Comparative:(evo_traj tum --ref gt_slam_UrbanNav-HK-TST-20210517.csv lio_odom_UrbanTST.tum -p):
[image]

When I try to use the "-a" command to put two trajectories together for comparison, My trajectory be like this
[image]

**I'm wondering if this is normal?**
In addition:
evo_ape tum gt_slam_UrbanNav-HK-TST-20210517.csv lio_odom_UrbanTST.tum -ap:
[image]

evo_ape tum gt_slam_UrbanNav-HK-TST-20210517.csv lio_odom_UrbanTST.tum -p:
[image]

# remove this line and paste your config HERE

```

MichaelGrupp, 2024-03-19:
If you add `--verbose` (or `-v`), you can see that this probably just caused by not all timestamps being matched to the other trajectory. So not a distortion, just missing data in between.

HowieHepburn, 2024-03-19:
Thank you very much for your reply, you are right, not all timestamps being matched to the other trajectory, and  I have solved the problem by adding t_max_diff=0.1 in my command