MichaelGrupp/evo #644: Evo crash when using evo_traj on ros bag file

Ab-Tx, 2024-03-19:
Hi,

**Description:**
Evo crashes when attempting to use evo_traj from a ROS bag file.
I am using evo from the terminal, no virtual environment.

**Command:**
```
evo_traj bag 2024-03-19-16-02-52.bag /path
```

**Console output:**
```

evo_traj bag 2024-03-19-16-02-52.bag /path
Traceback (most recent call last):
  File "/home/labrob/.local/lib/python3.8/site-packages/evo/entry_points.py", line 95, in launch
    main_module.run(args)
  File "/home/labrob/.local/lib/python3.8/site-packages/evo/main_traj.py", line 179, in run
    import evo.core.lie_algebra as lie
  File "/home/labrob/.local/lib/python3.8/site-packages/evo/core/lie_algebra.py", line 25, in <module>
    import scipy.spatial.transform as sst
  File "/usr/lib/python3/dist-packages/scipy/spatial/__init__.py", line 99, in <module>
    from .kdtree import *
  File "/usr/lib/python3/dist-packages/scipy/spatial/kdtree.py", line 8, in <module>
    import scipy.sparse
  File "/usr/lib/python3/dist-packages/scipy/sparse/__init__.py", line 229, in <module>
    from .base import *
  File "/usr/lib/python3/dist-packages/scipy/sparse/base.py", line 8, in <module>
    from .sputils import (isdense, isscalarlike, isintlike,
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <module>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <listcomp>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/home/labrob/.local/lib/python3.8/site-packages/numpy/__init__.py", line 320, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'

```

**Config:**
```
$ evo pkg --version
v1.26.2

$ evo pkg --pyversion
3.8.10

Ubuntu20.02

Source unchanged.

$ evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_enabled": true,
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```

(Archived ROS bag file)
[2024-03-19-16-02-52.bag.zip](https://github.com/MichaelGrupp/evo/files/14653742/2024-03-19-16-02-52.bag.zip)

MichaelGrupp, 2024-03-19:
I suspect that this is some issue with the particular scipy or numpy version that you use (searching the error message shows that other people also had similar issues with some versions). Because I can't reproduce this in a fresh virtual environment with `numpy==1.24.4` and `scipy==1.10.1`.

Which versions does `pip freeze | grep -e scipy -e numpy` show for you?

Ab-Tx, 2024-03-19:
It shows:
`$ pip freeze | grep -e scipy -e numpy
numpy==1.24.4
scipy==1.3.3
`

Ab-Tx, 2024-03-19:
After updating spicy 
` pip3 install scipy==1.10.0 `
I no longer have the error. Seems pip won't install the latest version of spicy when using "pip install spicy --upgrade".

Right now, evo_traj outputs an error claiming the message type is not supported.  Is this expected?
Console out:
```
$ evo_traj bag 2024-03-19-16-02-52.bag /path
[ERROR] unsupported message type: nav_msgs/msg/Path
```

MichaelGrupp, 2024-03-21:
> pip won't install the latest version of spicy when using `pip install spicy --upgrade`

"spicy" means something different :hot_pepper: ... what happens if you use **scipy**?

(looks like spicy is some sort of[ joke package](https://github.com/Carreau/spicy) that just lists scipy as dependency, but this won't upgrade scipy)

> `[ERROR] unsupported message type: nav_msgs/msg/Path`

As the message said, Path messages are not supported. And this probably won't change in the API/CLI, because Path topics have characteristics that would be a bit ugly to handle here (in a generalized way):

* each [message](http://docs.ros.org/en/noetic/api/nav_msgs/html/msg/Path.html) in a Path topic contains a full trajectory
* if the path topic contains more than one Path message, this means that there will be multiple trajectories to choose from
* this is completely different than in the other supported types that have a single trajectory per topic, so it would require a complete change of the user interface just for one message type --> won't do

closed 2024-03-21 by Ab-Tx

Ab-Tx, 2024-03-21:
The command "spicy" by itself does nothing:
```
$ Python3 spicy

Command 'Python3' not found, did you mean:

  command 'cython3' from deb cython3 (0.29.14-0.1ubuntu3)
  command 'python3' from deb python3 (3.8.2-0ubuntu2)

Try: sudo apt install <deb name>

$ python3 spicy
python3: can't open file 'spicy': [Errno 2] No such file or directory
```

> if the path topic contains more than one Path message, this means that there will be multiple trajectories to choose from
I'll attempt to republish the information I need in a **geometry_msgs/PoseStamped** message using a ros node and use that with evo.
Thank you for the help.