MichaelGrupp/evo #645: The ape is too large

huyongkang123, 2024-03-20:
**Description:**
I use evo_ape to evaluate the trajectory error of GPS and Liosam. The error appears to be significant at a certain position, but it doesn't appear to be that large. I don't know why.
The red box displays a significant positional error.
[image]

But from the picture, it doesn't seem that big.
[image]
But from the picture, it doesn't seem that big.
**Command:**
```
evo_ape tum ./RTK+IMU.txt ./LIO-SAM_20240312160111_witIMU_2.txt -p --plot_mode xy 
```

**Console output:**
```
APE w.r.t. translation part (m)
(not aligned)

       max      1.991285
      mean      0.457529
    median      0.366483
       min      0.001730
      rmse      0.639041
       sse      878.818432
       std      0.446139
```

**Additional files:**
Please attach all the files needed to reproduce the error.
[LIO-SAM_20240312160111_witIMU_2.txt](https://github.com/MichaelGrupp/evo/files/14661571/LIO-SAM_20240312160111_witIMU_2.txt)
[RTK+IMU.txt](https://github.com/MichaelGrupp/evo/files/14661574/RTK%2BIMU.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.25.1
* Python version shown by `evo pkg --pyversion`: 3.8.10
* operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu20.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
{
    "console_logging_forma
[RTK+IMU.txt](https://github.com/MichaelGrupp/evo/files/14661573/RTK%2BIMU.txt)
t": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_length_unit": "m",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "pygments_style": "monokai",
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}
```

MichaelGrupp, 2024-03-22:
If you enable `evo_config set plot_pose_correspondences true`, you can see which poses are matched and compared when zooming in the plot. The distance you see there explains your result. You need to check whether this is what you expect, given the data and time-sync.

[image]

huyongkang123, 2024-03-24:
I konw why，thank you！

closed 2024-03-24 by huyongkang123